xtr4nge/FruityWifi #77: Any had a successful ubuntu 14.04 install?

bytedeez, 2015-01-03:
Tried it but hasn't worked as of yet.

headii, 2015-01-04:
Hello bytedeez,

i receive the same error as you, when i try to install the modules on debian wheezy.

File "install-modules.py", line 69, in
if (float(version)) >= float(info['required']):
ValueError: invalid literal for float(): 2.1.2

If you change your /usr/share/fruitywifi/www/config/config.php from "2.1.2" to "2.12" then run the "install-modules.py"  it should run normal. 

Have a look @ https://github.com/xtr4nge/FruityWifi/issues/76

You can also install the modules over the webinterface, that should work fine!

xtr4nge, 2015-01-08:
Hi, thanks a lot for let me know.
I will update the script as soon as possible.

regards,
xtr4nge

xtr4nge, 2015-01-11:
install-modules.py script has been updated.

regards,
xtr4nge

closed 2015-01-11 by xtr4nge