xtr4nge/FruityWifi #78: Nethunter support

Zaxscdvf, 2015-01-07:
would be really cool to see this with nethunter, i was looking around and there seems to be problems with setting up. so if you guys could look into it that would be great.

xtr4nge, 2015-01-08:
Hi Zaxscdvf,
I hope you are well.
We are working together with Kali Linux team to include FruityWifi in Kali Linux repositories.
You will have more news soon ;)

regards,
xtr4nge

closed 2015-01-08 by xtr4nge

daforcebreda, 2015-01-09:
Hi xtr4nge,

first wish you the best for this fruityfull 2015. ;)

Hi i hope that there also is an automatic update future in fruitywifi,
so we can update it to the latest version. an update automatic the modules.

mabye an update future like what you can see on SME Server from
conntribs.org or something else.

not only on x86, but also on the ARM Platforn, like the raspberry pi.

can you make an image for me for the raspberry pi model B with the
ralink ty5370 chip.

or mabye help me with it, with a step for step installation.

now i get problems with loading the webinterface.
I ry it on kali linux, but also on latest wheezy.

Regards,

Paul

WilsonBradley, 2015-01-09:
Hi ,

I second that request for Kali on Raspberry distro and also the updating of
FruityWiFi
Please please please...

Sincerely,

Wilson Bradley
Wilsonb@pobox.com

Lan line & instant voice mail in USA (512)650-4599
My LinkedIn profile. http://lnkd.in/DiPBGk

xtr4nge, 2015-07-03:
Hi Guys
I will look into this ;)

regards,