xtr4nge/FruityWifi #79: 404 - Not found after updating modules

TAz00, 2015-01-08:
I installed FruityWifi on Kali 1.0.9a, and applied the fix to update modules (www/config/config.php -> Version = 2.12)

After running ./install-modules.sh and then rebooting, the server address has moved to https://127.0.1.1:8443/ instead of previous 127.0.0.1

root@kali:~# /etc/init.d/apache2 stop 
[....] Stopping web server: apache2apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1 for ServerName
. ok

xtr4nge, 2015-01-08:
Hi TAz00,
I will fix the problem with the float numbers as soon as possible. 
Note: the script does not change the IP address.

regards,
xtr4nge

xtr4nge, 2015-01-11:
install-modules.py script has been updated.

regards,
xtr4nge

closed 2015-01-11 by xtr4nge