xtr4nge/FruityWifi #80: Problem with Karma output

dimitrisre, 2015-01-12:
I have succesfully installed an use Karma.But i cannot see anything on the output area of the module. In fact not only for Karma. If i run Karma from command line i can see all the output.
Can you please help me? 
This is my configuration
[image]

xtr4nge, 2015-01-12:
Hi dimitrisre,
The log output from Karma it is not implemented in FruityWifi yet (if that is what you are looking for)
Try with [AP] Hostapd-Mana (You need to install Mana module to get this option)

regards,
xtr4nge

dimitrisre, 2015-01-12:
Ok! Thanks for the answer.

xtr4nge, 2015-01-17:
No prob. I will publish an update as soon as the Karma logs are implemented.

regards,
xtr4ge

closed 2015-01-17 by xtr4nge