xtr4nge/FruityWifi #81: No start Mana

lnkakalink, 2015-01-13:
Configuration file: /usr/share/fruitywifi/www/modules/mana/includes/conf/hostapd.conf
nl80211: Could not configure driver mode
nl80211 driver initialization failed.
hostapd_free_hapd_data: Interface wlan0 wasn't started
Configuration file: /usr/share/fruitywifi/www/modules/mana/includes/conf/hostapd.conf
nl80211: Could not configure driver mode
nl80211 driver initialization failed.
hostapd_free_hapd_data: Interface wlan0 wasn't started

xtr4nge, 2015-01-17:
Hi lnkakalink,
Please notice that not all chipsets are supported by Karma and Mana (for example Realtek chipsets)
What wireless card are you using? chipset?

regards,
xtr4nge

lnkakalink, 2015-01-17:
Hi xtr4nge,
D-Link DWA-125 

regards,
lnkakalink

xtr4nge, 2015-01-17:
Hi lnkakalink,
Looks like the D-Link DWA-125 has has a Ralink RT3070
You can get more details about you wireless card with the command: lsusb

They already reported that the RT3070 it is not supported by Hostapd (Karma and/or Mana)
`https://github.com/xtr4nge/FruityWifi/issues/61`

regards,
xtr4nge

closed 2015-01-17 by xtr4nge