xtr4nge/FruityWifi #83: Install FruityWifi x86 in Kali1.09,but can't connect to the server.

ghosttang, 2015-01-15:
Type "http://localhost:8000" in browser,but unable connect to the web server.
When I reload the ngnix, it announce :
=======================================================================
/usr/sbin/nginx -s reload

nginx: [emerg] SSL_CTX_use_certificate_chain_file("/etc/nginx/ssl/nginx.crt") failed (SSL: error:02001002:system library:fopen:No such file or directory error:20074002:BIO routines:FILE_CTRL:system lib error:140DC002:SSL routines:SSL_CTX_use_certificate_chain_file:system lib)

root@kali:~# ps -ef |grep ngnix

root      4200  4139  0 12:29 pts/1    00:00:00 grep ngnix
=======================================================================
How should I do now?

xtr4nge, 2015-01-17:
Hi ghosttang,
please verify that you have the ssl certificate and then restart nginx and php-fpm ;)

regards,
xtr4nge

ghosttang, 2015-01-18:
Yeah,it works!
Thank you.

regards,
ghosttang

xtr4nge, 2015-01-18:
Hi ghosttang,
No problem, if you have any other question, please let me know ;)

regards,
xtr4nge

closed 2015-01-18 by xtr4nge

exqua, 2015-11-12:
Hello xtr4nge, 
First of all. really nice project man!
But.. i have a small issue.
I can't connect wifi with any device. Wifi is up and running, SSID is visible.

Running on my Rasp B+ with v2.2 panel.

What can i do to get this working?

Greetings,
Exqua