xtr4nge/FruityWifi #84: dhcp doenst work

Asphyriax, 2015-01-18:
I have FruityWifi Installed on my Raspberry B+ but the dhcp server doenst work ??

xtr4nge, 2015-01-18:
Hi Asphyriax,
I need more details about your problem:

- What wireless card are you using? chipset?
- What [AP] mode are you using? (config page)
- Is it the wireless status > `enabled`?

regards,
xtr4nge

Asphyriax, 2015-01-18:
hello xtr4nge,

first of all: youre awesome,
second im kinda new in this so i will have some more questions.

i wanted to install the img but i dont know what my destination path was so i just installed it in Raspbian.

My wifi card is a ALFA USB WiFi AWUS036NEH
what are the modes;
IN - OUT | AP
IN - -- | AP
IN - OUT
IN - --
-- - OUT

i am formilliar with IP's,Masks, and gateways,

yes wireless is enabled
i was connected with my phone but doesnt reach the internet

i have tried them both but not with all modes

and i had all the modules installed but 3 of them worked

xtr4nge, 2015-01-20:
Hi Asphyriax
I hope you are well.

Looks like AWUS036NEH it is not supported by Hostapd, please check the following post: https://github.com/xtr4nge/FruityWifi/issues/61

But you still have the Airmon-ng option. Just go to the config page, and set the [AP] mode to `Airmon-ng`

regards,
xtr4nge

closed 2015-01-30 by xtr4nge