xtr4nge/FruityWifi #86: Can't start Mana

grayburn, 2015-01-21:
Did a clean install of Kali 1.09 on my Beaglebone Black. Cloned the FruityWifi repo and ran the install-FruityWifi.sh script. Install went fine. Logged into the web interface and installed most modules.

Using the TP-Link WN722N which is an atheros chipset
I switch the AP to Hosted-mana mode and try to start the Mana module, but it never stays enabled. From the wdsl page, when I try to start mana, it switches green/enabled, but quickly goes back to disabled. Also I can't seem to find any log files related to troubleshoot.

Do I need to start with the Master.zip and not cloning the repo?

xtr4nge, 2015-01-21:
Hi grayburn,
Do you have the same problem with Hostapd-Karma mode?

regards,
xtr4nge

grayburn, 2015-01-23:
Hi sorry for responding late. I worked on it a bit more and downloaded the master.zip and installed that over what I had. I am now able to get Karma working as well as mana.

closed 2015-01-23 by grayburn

xtr4nge, 2015-01-25:
No problem grayburn,
Good to know that you resolved the problem ;)

regards,
xtr4nge