xtr4nge/FruityWifi #87: Phishing module doesn't work (no redirecting)

FredTacker, 2015-01-21:
Hello,
I have a problem with the Phishing module. I connect to my FruityWifi but
When I start the Phishing module and try to go to facebook.com or. twitter.com the raspberry pi doesn't redirect me to the modified pages in /var/www/site/
Do you have any idea how to solve the problem?

My files in /var/www
index.php  site  tmp-squid
My files in /var/www/site/
1  2  3  4  5  data.txt  error.php  index.php  inject

To run http://10.0.0.1/index.php or  http://10.0.0.1/site/1/ in the browser is not a problem.
The browser is showing me the page when.

I hope you can help me.

xtr4nge, 2015-01-21:
Hi FredTacker,
You need to setup and enable DNSspoof + Phishing modules together.
DNSspoof module is what force the user to redirect to your phishing website ;)

regards,
xtr4nge

FredTacker, 2015-01-22:
okay I enabled DNSSpoof + Phishing now.
Are this  Hosts settings in DNSSpoof right?
10.0.0.1 example.com
10.0.0.1 facebook.com
10.0.0.1 gmail.com
10.0.0.1 twitter.com

xtr4nge, 2015-01-22:
Hi FredTacker,
Yes, that is correct. 
Note that depends on how the domain is requested will match or not (sometimes), example: 
`www.example.com` it is not the same as `example.com`

You can also try with [`*`] : `*.example.com`
Don't forget to check the IP of the webserver (`10.0.0.1` by default)

For more details, please search for `dnsspoof` man.

regards,
xtr4nge

FredTacker, 2015-01-22:
Hi xtr4nge thanks for your help.
I got it almost to work. But if i try to brows to facebook or twitter.com or gmail.com
the browser is saying this is a untrusted connection like this:
http://www.pic-upload.de/view-25920860/Screen-Shot-2015-01-22-at-17.45.53.png.html

The output of DNSSpoof looks like this:
10.0.0.75.55918 > 10.0.0.1.53:  49752+ A? www.facebook.com
10.0.0.57.60700 > 10.0.0.1.53:  30924+ A? www.facebook.com
10.0.0.57.53290 > 10.0.0.1.53:  16294+ A? twitter.com
10.0.0.57.55491 > 10.0.0.1.53:  38303+ A? www.gmail.com
10.0.0.57.60607 > 10.0.0.1.53:  13605+ A? www.facebook.com
10.0.0.57.58196 > 10.0.0.1.53:  12299+ PTR? 57.0.0.10.in-addr.arpa
10.0.0.57.53262 > 192.168.2.1.53:  43895+ PTR? 57.0.0.10.in-addr.arpa
dnsspoof: listening on wlan0 [udp dst port 53 and not src 10.0.0.1]

Does the (+ A?) means that the browser doesn't find the page?

Sorry for the circumstances.

xtr4nge, 2015-03-10:
It is not possible to bypass the certificate error with some sites, as the certificate that you are using is a self signed certificate and also does not match the domain that you are spoofing. For some websites you can try combining Phishing, DNSspoof and SSLstrip.

regards,
xtr4nge

closed 2015-03-10 by xtr4nge