xtr4nge/FruityWifi #88: Modes

Asphyriax, 2015-01-22:
Hello, 

What are the modes;
IN - OUT | AP
IN - -- | AP
IN - OUT
IN - --
-- - OUT

I am new in this FruityWifi and was wondering what all the modes are.

Greetings Asphyriax,

WilsonBradley, 2015-01-22:
This was also not clear to me either.

Sincerely,

Wilson

Asphyriax, 2015-01-22:
Hello Wilson,

On what device or pc are you trying to install it??

If somebody would make a video with the explanation of al those modes and how to make it work because input and output have a different meaning i suppose.

Greets Asphyriax,

AnguisCaptor, 2015-03-11:
IN = Incoming connections allowed
OUT = Outgoing connections allowed

Obviously you'd need to have IN, if you want people to have internet aswell use OUT
So if you want people connecting to your AP simply use IN - OUT | AP

Asphyriax, 2015-10-19:
im sorry for my stupidity, with the time i learn more and more about linux,

i have a raspberry pi b+ with raspian wheezy and the FW installed with a TP-LINK TL-WN722N

i am currently trying to get the whatsapp module up and running, it wil run but won't find any numbers, do i need to run something else with it?

xtr4nge, 2015-10-19:
HI Asphyriax,
For debugging, you can run the script manually without using FruityWiFi web-interface:

`cd /usr/share/fruitywifi/www/modules/whatsapp/includes/`
`./whatsapp_discover`

For further details, please post the question on the forum:
http://fruitywifi.boards.net

regards,

closed 2015-10-19 by xtr4nge

Asphyriax, 2015-10-19:
tried that one Xtr4nge but i have no clue why it does't capture phone numbers? am i missing out on something? maybe another module?

thx,

also good work and nice coding:)

AnguisCaptor, 2015-10-19:
You're not able to grab phonenumbers from Android phones as they're
encrypting the Whatsapp traffic
You are able to grab iPhone numbers because they're not encrypting traffic
I guess you're trying it with a android phone which will no longer work,
try iPhone ;)

Asphyriax, 2015-10-19:
whatsapp is also encrypted on iphone, probely are the recent updates en stuff.

im going to try to get this crap working http://www.claudiodinardo.com/arp-spoofing-mitm-attack-over-ssl-splitted-connections-and-fake-ca-certificate-forging-all-together-against-an-iphone/ this is probaly the problem,