xtr4nge/FruityWifi #89: Error web interface - no login possible

johnjdoe, 2015-01-29:
After installing FruityWifi as described in the Wiki (https://github.com/xtr4nge/FruityWifi/wiki/Install) on a machine with a Tanglu (Debian) distrib, I receive the following when I logged in on http://localhost:8000 (admin/admin):

. */ ?> "; if ($users[$user] == md5($pass)) { $_SESSION["user_id"] = $user; //echo "welcome"; header('Location: page_status.php'); } else { //echo "get out of here..."; header('Location: index.php?error=1'); } ?>

I saw already the following signs before the FruityWifi logo on the logon screen: . */ ?>

I already tried to reboot the machine.

Any help is apreciated! Thank you in advance!

johnjdoe, 2015-01-30:
I found the problem/solution:

It seems that in a default installation of PHP-FPM in Debian Jessie PHP's short opening tags is disabled.

As the PHP code of FruityWifi always start with "<?" instead of "<?PHP" it is not interpreted/executed.

It worked fine after enabling short opening tags in PHP.INI:

short_open_tag = On

Maybe it's a good idea to change all "<?" to "<?PHP" in the code?

xtr4nge, 2015-01-30:
Hi johnjdoe,
I hope you are well.

It has seen before the problem with the short open tags.
It is not planned (at least for the moment) to change the short open tags to `<?PHP`. 
The setup requires to have `short_open_tag = On`
Thanks a lot for the update ;)

regards,
xtr4nge

closed 2015-01-30 by xtr4nge