xtr4nge/FruityWifi #90: fruity php error / karma and mana don´t work

freakywhezzl, 2015-02-02:
Hey,
i´ve install fruitwifi under "kali-linux-1.0.9a-i386" on my laptop.

when i start fruity - the error log is full of php error´s:

`2015/02/02 15:26:02 [error] 9094#0: *1 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined index: error in /usr/share/fruitywifi/www/index.php on line 68" while reading response header from upstream, client: 127.0.0.1, server: , request: "GET / HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8000"
2015/02/02 15:26:06 [error] 9094#0: *1 FastCGI sent in stderr: "PHP message: PHP Notice:  A session had already been started - ignoring session_start() in /usr/share/fruitywifi/www/login_check.php on line 20
PHP message: PHP Notice:  Undefined index: service in /usr/share/fruitywifi/www/page_status.php on line 87
PHP message: PHP Notice:  Undefined index: action in /usr/share/fruitywifi/www/page_status.php on line 88
PHP message: PHP Notice:  Undefined index: service in /usr/share/fruitywifi/www/page_status.php on line 90
PHP message: PHP Notice:  Undefined index: action in /usr/share/fruitywifi/www/page_status.php on line 91" while reading response header from upstream, client: 127.0.0.1, server: , request: "GET /page_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8000", referrer: "http://localhost:8000/"
2015/02/02 15:26:07 [error] 9094#0: *1 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined index: reveal_public_ip in /usr/share/fruitywifi/www/page_status.php on line 323
PHP message: PHP Notice:  Undefined variable: data in /usr/share/fruitywifi/www/page_status.php on line 353
PHP message: PHP Notice:  Undefined offset: 2 in /usr/share/fruitywifi/www/page_status.php on line 357
PHP message: PHP Notice:  Undefined offset: 1 in /usr/share/fruitywifi/www/page_status.php on line 357
PHP message: PHP Notice:  Undefined offset: 3 in /usr/share/fruitywifi/www/page_status.php on line 357" while reading upstream, client: 127.0.0.1, server: , request: "GET /page_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8000", referrer: "http://localhost:8000/"
2015/02/02 15:26:09 [error] 9094#0: *1 FastCGI sent in stderr: "PHP message: PHP Notice:  A session had already been started - ignoring session_start() in /usr/share/fruitywifi/www/login_check.php on line 20
PHP message: PHP Notice:  Undefined variable: data in /usr/share/fruitywifi/www/page_logs.php on line 47
PHP message: PHP Notice:  Undefined variable: data in /usr/share/fruitywifi/www/page_logs.php on line 47
PHP message: PHP Notice:  Undefined variable: data in /usr/share/fruitywifi/www/page_logs.php on line 47" while reading response header from upstream, client: 127.0.0.1, server: , request: "GET /page_logs.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8000", referrer: "http://localhost:8000/page_status.php"
2015/02/02 15:26:09 [error] 9094#0: *1 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: data in /usr/share/fruitywifi/www/page_logs.php on line 47
PHP message: PHP Warning:  fopen(/var/log/fruitywifi/sudo.log): failed to open stream: Permission denied in /usr/share/fruitywifi/www/page_logs.php on line 33
PHP message: PHP Warning:  fread() expects parameter 1 to be resource, boolean given in /usr/share/fruitywifi/www/page_logs.php on line 35
PHP message: PHP Warning:  fclose() expects parameter 1 to be resource, boolean given in /usr/share/fruitywifi/www/page_logs.php on line 36" while reading upstream, client: 127.0.0.1, server: , request: "GET /page_logs.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8000", referrer: "http://localhost:8000/page_status.php"
2015/02/02 15:26:38 [error] 9094#0: *1 FastCGI sent in stderr: "PHP message: PHP Notice:  A session had already been started - ignoring session_start() in /usr/share/fruitywifi/www/login_check.php on line 20
PHP message: PHP Notice:  Undefined index: filename in /usr/share/fruitywifi/www/page_config_adv.php on line 38
PHP message: PHP Notice:  Undefined index: newdata in /usr/share/fruitywifi/www/page_config_adv.php on line 39
PHP message: PHP Notice:  Undefined index: iface in /usr/share/fruitywifi/www/page_config_adv.php on line 40
PHP message: PHP Notice:  Undefined index: io_out_iface in /usr/share/fruitywifi/www/page_config_adv.php on line 41
PHP message: PHP Notice:  Undefined index: io_in_iface in /usr/share/fruitywifi/www/page_config_adv.php on line 42
PHP message: PHP Notice:  Undefined index: iface_supplicant in /usr/share/fruitywifi/www/page_config_adv.php on line 43
PHP message: PHP Notice:  Undefined index: newSSID in /usr/share/fruitywifi/www/page_config_adv.php on line 44
PHP message: PHP Notice:  Undefined index: hostapd_secure in /usr/share/fruitywifi/www/page_config_adv.php on line 45
PHP message: PHP Notice:  Undefined index: hostapd_wpa_passphrase in /usr/share/fruitywifi/www/page_config_adv.php on line 46
PHP message: PHP Notice:  Undefined index: supplicant_ssid in /usr/share/fruitywifi/www/page_config_adv.php on line 47
PHP message: PHP Notice:  Undefined index: supplicant_psk in /usr/share/fruitywifi/www/page_config_adv.php on line 48
PHP message: PHP Notice:  Undefined index: pass_old in /usr/share/fruitywifi/www/page_config_adv.php on line 49
PHP message: PHP Notice:  Undefined index: pass_new in /usr/share/fruitywifi/www/page_config_adv.php on line 50
PHP message: PHP Notice:  Undefined index: pass_new_repeat in /usr/share/fruitywifi/www/page_config_adv.php on line 51
PHP message: PHP Notice:  Undefined index: service in /usr/share/fruitywifi/www/page_config_adv.php on line 52
PHP message: PHP Notice:  Undefined index: action in `

.... and so on.

when i start wireless with karma the karma log says:

`Configuration file: /usr/share/fruitywifi/www/modules/karma/includes/conf/hostapd.conf
Failed to update rate sets in kernel module
Could not connect to kernel driver.
Using interface wlan0 with hwaddr 74:e5:0b:c6:ac:42 and ssid 'FruityWifi'
Failed to set beacon parameters
Could not connect to kernel driver.  `

My wifi driver is "nl80211".

What can i do to start karma or mana?

hostapd.conf:

`ctrl_interface=/var/run/hostapd-phy0
driver=nl80211
wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_txop_limit=0
wmm_ac_be_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0
tx_queue_data3_aifs=7
tx_queue_data3_cwmin=15
tx_queue_data3_cwmax=1023
tx_queue_data3_burst=0
tx_queue_data2_aifs=3
tx_queue_data2_cwmin=15
tx_queue_data2_cwmax=63
tx_queue_data2_burst=0
tx_queue_data1_aifs=1
tx_queue_data1_cwmin=7
tx_queue_data1_cwmax=15
tx_queue_data1_burst=3.0
tx_queue_data0_aifs=1
tx_queue_data0_cwmin=3
tx_queue_data0_cwmax=7
tx_queue_data0_burst=1.5
hw_mode=g
channel=6

logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
#ieee80211n=1
#ht_capab=[HT20][SHORT-GI-20][SHORT-GI-40][RX-STBC1][DSSS_CCK-40]

interface=wlan0
ctrl_interface=/var/run/hostapd-phy0
disassoc_low_ack=1
auth_algs=1
wpa=0
ssid=FruityWifi
#bridge=br-lan
wmm_enabled=1
bssid=74:e5:0b:c6:ac:42
ignore_broadcast_ssid=0  `

thx & greetz

whezzl

freakywhezzl, 2015-02-02:
hey.

when i install "fruitywifi" over the installscript from the github.com page.
i dont get an error.log area with the php errors.

when i start the wireless card with karma or mana ap mode, i still get the errors on karma logfile:
```
Configuration file: /usr/share/fruitywifi/www/modules/karma/includes/conf/hostapd.conf
Failed to update rate sets in kernel module
Could not connect to kernel driver.
Using interface wlan0 with hwaddr 74:e5:0b:c6:ac:42 and ssid 'FruityWifi'
Failed to set beacon parameters
Could not connect to kernel driver.
Configuration file: /usr/share/fruitywifi/www/modules/karma/includes/conf/hostapd.conf
Failed to update rate sets in kernel module
Could not connect to kernel driver.
Using interface wlan0 with hwaddr 74:e5:0b:c6:ac:42 and ssid 'FruityWifi'
Failed to set beacon parameters
Could not connect to kernel driver.
```

My network kernel driver is:

```
08:00.0 Network controller: Intel Corporation Centrino Wireless-N 1000 [Condor Peak]
	Subsystem: Intel Corporation Centrino Wireless-N 1000 BGN
	Kernel driver in use: iwlwifi
```

###### Is this driver supportet for karma or mana?

greetz whezzl

xtr4nge, 2015-02-02:
Hi freakywhezzl,
Realtek chipset (and some others) are not supported by `Hostapd` so does not work with `Karma` and/or `Mana` [AP] mode.
You can only use [AP] mode Airmon-ng with Realtek chipsets, but some modules will not work (like Karma and/or Mana).
I recommend to use Atheros chipsets.

regards,
xtr4nge

closed 2015-02-02 by xtr4nge