xtr4nge/FruityWifi #92: Autostart Karma (Monitor mode)

AnguisCaptor, 2015-02-07:
Heey what's up, thanks for your outstanding work :D
I'm having a bit of a problem, I'm running the raspberry pi version 1.2.1 if I'm not misstaken (lastest of today)

And when ever I put Karma to the Autostart and enable it... it doesn't start karma
I think it's a problem of not having Monitor Mode enabled before karma

Would you be so kind enough to add a option to enable monitor mode at the AutoStart menu
So we can all enjoy Karma at boot :)

xtr4nge, 2015-02-08:
Hi AnguisCaptor,
I hope you are well. Thanks a lot for letting me know, I will take a look ;)

regards,
xtr4nge

AnguisCaptor, 2015-02-10:
Cool, I will take a look aswell, I am not a "web dev" myself as I simply don't like it, but I will try to contribute to the project :)

xtr4nge, 2015-08-30:
Module has been updated.
Thanks AnguisCaptor ;)

regards,

closed 2015-08-30 by xtr4nge