xtr4nge/FruityWifi #95: Raspberry Pi 2

MrBeanzy, 2015-02-16:
Has anyone tried FruityWifi on the Raspberry Pi 2? My old pi is fine, but the new one wont start the wireless?

xtr4nge, 2015-02-16:
Hi MrBeanzy,
Yes, I already tested it on RPI2 and it works great!
But with RPIv1 it also works fine. 
What is the problem that you are having?

Please note that the wireless card needs to be supported by Hostapd, if not, you will need to use the Airmon-ng mode. (check the config page on FruityWifi interface)

Please let me know ;)
regards,
xtr4nge

MrBeanzy, 2015-02-16:
Thanks for the quick reply xtr4nge. I have setup a clean install of raspbian, downloaded the latest FruityWiFi master zip and installed using the script. But every time i try to start the WiFi it just says enabled for a few seconds and then goes back to disabled. It also looks like when i click to install modules,  it says downloading but the modules never appear after? Also if i change anything on the config page it changes back immediatly to the defaults.  I have tried with a Realtek 8188 and a Atheros AR9002U. They both work if i manually setup the wireless cards as access points? Any pointers would be brilliant. Thanks!

MrBeanzy, 2015-02-16:
Completely my fault! Permissions were wrong! Thanks for your help xtr4nge. Works very well on RP2! Very fast

closed 2015-02-21 by MrBeanzy

KeKkers, 2015-02-24:
Could you elaborate on the permissions solution please? having the same issue.... Thanks,

xtr4nge, 2015-02-27:
Hi MrBeanzy,
Thanks for the update ;)

regards,
xtr4nge

materlo, 2015-04-04:
Hi MrBeanzy, you can say like you fixed the problem please? 

I can not enable wireless