xtr4nge/FruityWifi #96: Captive Portal

MrBeanzy, 2015-02-21:
After installing the captive portal and enabling it i was expecting to see a page on my device asking for my email address? But i get forwarded to a 404 page? I take it the files may need copying to the root of the web server? 

Thanks for all the help so far, Paul.

MrBeanzy, 2015-02-21:
Also it appears if you browse a website with https, you get a certificate error?

xtr4nge, 2015-03-10:
Hi MrBeanzy,
When Captive Portal is enable in FruityWifi, it creates a softl-ink into /var/www/ and it adds a line inside /var/www/index.php to force the redirect. You can combine Captive Portal with DNSSpoof for example.

You get the certificate message because it is a self signed certificate. You can get and config a proper certificate if you want. 

regards,
xtr4nge

closed 2015-03-10 by xtr4nge