xtr4nge/FruityWifi #98: No Wifi Or Modules

KeKkers, 2015-02-25:
Hello, still quite new to linux, slowly getting my head around it (why i bought a PI!). Im struggling to get any where with FruityWifi. Installed fine, login fine, but i dont have any modules and cant install them, just get a downloading screen and thats it!
Also if i try and turn on wireless it turns straight off, if i change any settings them also instantly default back.
Any ideas?
Thanks,

KeKkers, 2015-02-26:
Sorted this. I had to assign the www-data user sudo manually and it worked. Is the install file supposed to do this automatically?

MrMister2005, 2015-02-26:
FruityWifi is GOOD  ... ok guy

KeKkers, 2015-02-27:
?!?!?

closed 2015-02-27 by KeKkers

xtr4nge, 2015-02-27:
Hi KeKkers,
You can find in the installation folder a script that the name is install-modules.py. You can use that script to install all modules. Please notice that the installation on RPI will take some time as some modules requires to compile tools.

About FruityWifi setup, please notice that not all wireless cards are supported by Hostapd (like Realtek chipsets), so depends on you wireless card you will needs to use the option [AP] Airmon-ng. 

regards,
xtr4nge