xtr4nge/FruityWifi #99: Accessing webpage for FruityWifi

joeyj2468, 2015-03-07:
Hello,

I have installed FruityWifi on my raspberry pi B+ for Kali linux. However, it will not let me access the webpage for fruityWifi.  I tried the http://localhost:80 and 8443 .  I followed the steps on your page:

https://github.com/xtr4nge/FruityWifi/wiki/Nginx

I am remoting into my raspberry pi with putty and I am trying to access the webpage on my windows computer.  Not sure If I am doing it wrong. Maybe ports are closed.  Please assist.  Thanks!

-Joe

xtr4nge, 2015-03-10:
Hi joeyj2468 
Check what are the open ports (for example using nmap). You need to have ports 8000 and 8443 open (those are the ports used by FruityWifi)

regards,
xtr4nge

joeyj2468, 2015-03-10:
I actually got it working yesterday. I did everything as said on your page, but ports were closed.  I reimaged the card and got it to work finally.  Now i am trying to get wifi adaptor to work.  It associates but disassociates with a local reason:3 deauth_failed.  It is a power issue, bit not sure how to fox it,  forums mention it, but not on how to solve it directly.  Any advice would help. Thanks again!

xtr4nge, 2015-03-10:
Hi joeyj2468,
The power issue it is solve using a powered usb hub for the wireless dongle ;)

regards,
xtr4nge

joeyj2468, 2015-03-10:
Interesting.  I read that somewhere. But its weird, since i tested wifi on my raspberry last month and it worked like a charm.  But after reimagining kali, it fails ti authenticate.  I will look into a few power hubs that work with raspberry pi.  Thanks man ;)

xtr4nge, 2015-03-10:
No prob ;)

regards,
xtr4nge

closed 2015-03-10 by xtr4nge