xtr4nge/FruityWifi #102: Unable to download modules & Pages are not responding correctly

AnguisCaptor, 2015-03-11:
Heey xtr4nge, thanks for helping earlier with the tag problem, that worked.

Now I've got a issue where the modules page is not working proplerly
I click at the tar.gz option and I try to click all the "Install" links, the popup comes up for like 100 milliseconds and dissapears, nothing happens after

And the config page does not respond, it simply refreshes on whatever I press/set and sets back the old value

I am pretty sure the 2 problems are having the same issue

jmmalcala, 2015-03-12:
There is an issue with root privileges with fruitywifi user.
Add with visudo command this line: fruitywifi	ALL=(ALL:ALL) NOPASSWD:ALL
Below root line.

AnguisCaptor, 2015-03-12:
Tried this but the problem is there there

xtr4nge, 2015-03-12:
Hi AnguisCaptor,
I hope you are well.
Please check if you have installed `git` as some people already reported this problem when `git` it is not installed. (`git ` is used to download the modules from the repositories.)
You can also download manually each of the modules and extract them in: /usr/share/fruitywifi/www/modules/
example: `usr/share/fruitywifi/www/modules/dnsspoof`

Please let me know if that solved the problem ;)

regards,
xtr4nge

AnguisCaptor, 2015-03-12:
Heey xtr4nge, I have git installed, currently I am able to install the modules, start and isntall them aswell
Currently the only problem I have left is the strange thing going on with the  "config" page
To actually select a item now with saving it to select the item 3x and hit save

This is what is going on (Happens with all the options in the Config page):
For example, I have the "Monitor" option in the additional Interfaces set currently on wlan0
When I want to change that to Wlan1, it refreshes the Config page and it's Wlan0 again
I try changing it again to Wlan1 and it's once again Wlan0
I try to change the option again to Wlan1 and now it finally sticks

This happens with all the dropbox items

xtr4nge, 2015-03-12:
Hi AnguisCaptor,
That sounds like a permissions problem.

Please check if you have the following file: `/etc/sudoers.d/fruitywifi`

With content:
`fruitywifi	ALL=(ALL:ALL) NOPASSWD:ALL`
`Defaults loglinelen = 0`
`Defaults logfile = /usr/share/fruitywifi/logs/sudo.log`
`Defaults log_year`
`Defaults log_host`
`Defaults syslog=auth`

FruityWifi execs all the commands using `sudo` as `fruitywifi` user.

regards,
xtr4nge

AnguisCaptor, 2015-03-12:
I checked the file and the same content is in that file like you showed as sample

xtr4nge, 2015-03-12:
Hi, sorry I misunderstood =p
Please check that you are using `page_config_adv.php` (from the top menu) and not `page_config.php` as it was deprecated and needs to be removed.

Please let me know ;)

regards,
xtr4nge

AnguisCaptor, 2015-03-12:
It's using page_config_adv.php

xtr4nge, 2015-03-12:
Hi AnguisCaptor,
I think I'm not understanding what is the problem that you are experiencing. 
Can you provide me some screenshot?
Are you able to install/remove modules using the FruityWifi interface?

regards,
xtr4nge

AnguisCaptor, 2015-03-12:
It's hard to give a screenshot as it happens quick, I will provide a gif or vid lateron to show what is happening

xtr4nge, 2015-03-12:
Cool, thanks a lot ;)

regards,
xtr4nge

AnguisCaptor, 2015-03-12:
Here you'd be able to see what is happening https://www.youtube.com/watch?v=dNGC3YE0ICM&feature=youtu.be

xtr4nge, 2015-03-13:
Hi AnguisCaptor,
Thanks a lot for the video. That is a very unusual behavior.

You can debug the posted variables adding some code to: `/usr/share/fruitywifi/www/scripts/config_iface.php`

try adding the following code after both includes at the top of the file:

`echo $_POST["ap_mode"];`
`exit;`

The output needs to be:
Hostapd: `1`
Hostapd-Mana: `3`
Hostapd-Karma: `4`
Airmon-ng: `2`

You can also debug what is going on in the background (remove first the code that I suggested before) with this line:
`tail -F /usr/share/fruitywifi/logs/sudo.log`
Then you can exec each of the commands that you get in the sudo.log manually to see what is failing.

Please let me know

regards,
xtr4nge

AnguisCaptor, 2015-03-13:
I went from Top to Bottom with the modes
The tail log: http://pastebin.com/hktfc2yV

Adding the echo did not much, no message shown not even with " echo 'test'; "

janukss, 2015-04-06:
I was having the same problem on a Raspberry Pi. Fixed it with:
`sudo chown root:root /etc/sudoers.d/fruitywifi`

closed 2015-04-25 by xtr4nge