xtr4nge/FruityWifi #105: Cannot start wireless

adrianTNT, 2015-03-17:
I installed ARM version (Raspberry Pi) with script: install-FruityWifi.sh and wifi dongle is Edimax ew-7811un

While install ran, I noticed that it installed nginx even if Apache2 was already running, this might be part of my problem(s).

Now in order to view the [ip]:8000 interface I need to manually stop Apache and start Nginx.

If I go to "status" in top menu and click "start" next to wireless, it doesn't start. Sometimes it shows the "loading fruit" text and sometimes it doesn't, it always returns to same page and it shows wireless as stopped. 

I tried changing [AP] from Hostapd to Airmon in config page but it didn't help.

If I go to "wsdl" page and click "edit" next to wireless, it goes to page_config.php , I read that the config page is old and it should go to page_config_adv.php (same url linked from "config" in top menu), if true this makes me think the program has many bugs and maybe I should stop trying.

That /etc/sudoers.d/fruitywifi file also contains the recommended lines. 

I think I ran the install script as root, was that the problem?

candrewneal, 2015-03-18:
I'm having the same issue as well. I've tried it on my Raspberry Pi B and my Raspberry Pi 2. I've also used the Kali Linux and the Raspian install methods on both Pi's. I've had no luck fixing the issue.

xtr4nge, 2015-03-18:
Hi,
You can have conflicts between Apache and Nginx

`/etc/init.d/apache2 stop`
`/etc/init.d/nginx start`
`/etc/init.d/php5-fpm start`

FruityWifi interface can be accessed from ports 8000 (http) and 8443 (https)

Please notice that not all wireless cards are supported, and with raspberry pi and some USB dongles a powered usb hub is required.
If your wireless card is not supported by Hostapd, you can check if your card + airmon-ng works with the following command:

`airmon-ng start wlan0`

if an iface mon0 is created (check with ifconfig), then it will work with FruityWifi + Airmon-ng option.

Any Atheros chipset works fine with FruityWifi, also for raspberry pi + FruityWifi, you can use: Ralink RT5370 (powered USB hub not required) ;)

Regarding the connectivity problem, I suggest to do the following change in /etc/network/interfaces: (to avoid interfaces conflict with FruityWifi)

`auto lo`
`iface lo inet loopback`
`allow-hotplug eth0`
`iface eth0 inet dhcp`
`#allow-hotplug wlan0`
`#iface wlan0 inet manual`
`#wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf`
`#iface default inet dhcp`

With Kali Linux you can install FrutiyWifi using `apt-get install fruitywifi`

Please let me know if that solved the problem.
regards,
xtr4nge

candrewneal, 2015-03-19:
Hey thanks for the reply on my issue. The problem was the wireless adapter I was using. I tried it out on my Alfa card on airmon-ng mode and I was able to broadcast the access point. However I tried connecting my Galaxy S3 to it and it wouldn't get a connection. I looked in the logs and it said no usable IP range or something like that. Thanks again for the help.

adrianTNT, 2015-03-19:
@candrewneal  try to discuss this in a separate topic because some of your issues are different than my initial ones and it will get offtopic.

candrewneal, 2015-03-19:
@adrianTNT Sorry about that...

xtr4nge, 2015-04-05:
Hi adrianTNT,
Has been this problem resolved?

regards,
xtr4nge

adrianTNT, 2015-04-05:
No, I installed other OS after these issues. But as I rememver Kali Linux also didn't detect this device so this could be a driver issue.

yoerithor, 2018-05-09:
Hey, if I install fruitywifi on raspbian os, it won’t start