xtr4nge/FruityWifi #106: nginx test failed

joeyj2468, 2015-03-17:
I am trying to open the ports, but when I do they are closed.  I tried to commands you gave me before but it will not start automatically upon reboot.  I tried /etc/init.d/nginx start and /etc/init.d/php5-fpm start.  Nginx keeps saying test failed while php seems to be ok.  I also tried to fix the nginx issue by going to your page and following the commands.  It seems I have the most updated nginx on my raspberry pi.  Also, when I click on a module like nmap, it will bring me to nginx page that gives its version.  I think it was 1.2.1 or 2.1.  Please advise.  thanks. 

[image]

joeyj2468, 2015-03-17:
When I click on nmap, I get this error.  I was able to fix the nginx.  Had a duplicate file.  Here is the error.

[image]

xtr4nge, 2015-03-18:
Hi,
You have a conflict in Nginx config files. Remove the duplicated entries or uninstall, clean the folder and reinstall.
Please notice that with Kali Linux you don't need to use the FruityWifi installation script, you just need to do `apt-get install fruitywifi`

regards,
xtr4nge

joeyj2468, 2015-03-18:
Hello,

Thanks for your advice.  Sorry if this sounds silly, but by clean do you mean remove the enitre contents of the NGINX folder? Another thing, can you explain Hostapd and how it is used with FruityWifi?  Still getting the hang of this.  Thanks.

xtr4nge, 2015-03-19:
Hi joeyj2468,
No prob. With the duplicate entries I mean to search inside the Nginx config files and remove any line/config that is creating the problem. The second option is to uninstall Nginx and (just in case) remove any file in the Nginx config folder.

FruityWifi can use 3 versions of Hostapd. By default it uses Hostapd from the repositories (apt-get -y install hostapd) but it can use 2 more versions, Hostapd-Mana and Hostapd-Karma, but you need to install the modules first. 
To use Hostapd the wireless card needs to be supported (for example Realtek chipsets are not supported).
As an alternative for not supported chipsets, FruityWifi implemented the option Airmon-ng.

Keep in mind that Raspberry Pi could have problems with some USB dongles, and a powered USB hub could be required.

regards,
xtr4nge

closed 2015-04-05 by xtr4nge