xtr4nge/FruityWifi #108: sslstrip, ngrep module failure on Kali Linux

killbop, 2015-04-05:
I've installed FruityWiFi from Kali repository.
Running latest version
```
uname -a
Linux kali 3.18.0-kali3-amd64 #1 SMP Debian 3.18.6-1~kali2 (2015-03-02) x86_64 GNU/Linux
```
Upon starting the sslstrip module it goes green for a few seconds and then goes down.
Module details page show

SSLstrip disabled. | start
Inject enabled.  | stop
Tamperer enabled. | stop

Same for ngrep and whatsapp - goes up for a brief and fails.

module logs are blank,
but errors.log is full off
cheers.
```
2015/04/02 22:36:35 [error] 7263#0: *57 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/supplicant/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:35 [error] 7263#0: *56 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/autostart/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:35 [error] 7263#0: *51 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/autossh/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:35 [error] 7263#0: *53 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/3g_4g/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:35 [error] 7263#0: *55 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/karma/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:35 [error] 7263#0: *365 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/mana/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:27 [error] 7263#0: *51 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/whatsapp/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:27 [error] 7263#0: *57 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: output in /usr/share/fruitywifi/www/scripts/status_logs_stations.php on line 16" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/status_logs_stations.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
PHP message: PHP Notice:  Undefined offset: 3 in /usr/share/fruitywifi/www/scripts/status_logs_dhcp.php on line 14" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/status_logs_dhcp.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
PHP message: PHP Notice:  Undefined offset: 1 in /usr/share/fruitywifi/www/scripts/status_logs_dhcp.php on line 14
2015/04/02 22:36:27 [error] 7263#0: *365 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined offset: 2 in /usr/share/fruitywifi/www/scripts/status_logs_dhcp.php on line 14
2015/04/02 22:36:27 [error] 7263#0: *55 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/tcpdump/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:27 [error] 7263#0: *53 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/urlsnarf/_info_.php on line 6" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:27 [error] 7263#0: *56 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/sslstrip/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:27 [error] 7263#0: *57 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/squid3/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:27 [error] 7263#0: *365 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/rpitwit/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:27 [error] 7263#0: *51 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/responder/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:27 [error] 7263#0: *53 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/ngrep/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
2015/04/02 22:36:27 [error] 7263#0: *55 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined variable: log_path in /usr/share/fruitywifi/www/modules/phishing/_info_.php on line 5" while reading response header from upstream, client: 127.0.0.1, server: , request: "POST /scripts/module_status.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php5-fpm-8000.sock:", host: "localhost:8443", referrer: "https://localhost:8443/page_status_wsdl.php"
[... cut ...]
PHP message: PHP Notice:  Undefined index: service in /usr/share/fruitywifi/www/page_status.php on line 90
PHP message: PHP Notice:  Undefined index: action in /usr/share/fruitywifi/www/page_status.php on line 88
PHP message: PHP Notice:  Undefined index: service in /usr/share/fruitywifi/www/page_status.php on line 87
2015/04/02 22:32:16 [error] 7263#0: *22 FastCGI sent in stderr: "PHP message: PHP Notice:  A session had already been started - ignoring session_start() in /usr/share/fruitywifi/www/login_check.php on line 20
PHP message: PHP Notice:  Undefin

xtr4nge, 2015-04-05:
Hi killbop,
I suspect that something is wrong with the installation or wireless (FruityWifi service) it is not enabled.
** Note that the modules require to have wireless (service) up and running (enabled).

SSLstrip works straight forward if the module is installed correctly. Please try the following command:
`/usr/share/fruitywifi/www/modules/sslstrip/includes/sslstrip -a -s -l 10000`
If SSLstrip is giving you an error, there is a problem with the installation. Uninstall and Install the module again.

The same with Ngrep, Please try the following command manually:
`/usr/bin/ngrep -q -d wlan0 -W byline  -t 'Cookie' 'tcp and port 80'`
(Note that wlan0 needs to be the name of the interface that you are using for the [AP] mode.)

And the same with Whatsapp module, please try the following command:
`/usr/share/fruitywifi/www/modules/whatsapp/includes/whatsapp_discover -i wlan0`
(Same as Ngrep, note that wlan0 needs to be the name of the interface that you are using for the [AP] mode.)

Let me know ;)

regards,
xtr4nge

killbop, 2015-04-05:
hey mate.
so.
```
/# /usr/share/fruitywifi/www/modules/sslstrip/includes/sslstrip -a -s -l 10000
bash: /usr/share/fruitywifi/www/modules/sslstrip/includes/sslstrip: Permission denied
```
but,
```
/usr/share/fruitywifi/www/modules/sslstrip/includes# sslstrip -a -s -l 10000

sslstrip 0.9 by Moxie Marlinspike running...
```

```
 /usr/bin/ngrep -q -d wlan0 -W byline -t 'Cookie' 'tcp and port 80'
wlan0: no IPv4 address assigned: Cannot assign requested address
interface: wlan0
filter: (ip or ip6) and ( tcp and port 80 )
match: Cookie
```

```
 /usr/share/fruitywifi/www/modules/whatsapp/includes/whatsapp_discover -i wlan0

Whatsapp Discover v1.0  --- Deepak Daswani (@dipudaswani) 2014
                            http://deepakdaswani.es 

Realtime sniffer  v1.0  --- Fork by xtr4nge (@xtr4nge) 2014
                            https://github.com/xtr4nge/whatsapp_discover
```

killbop, 2015-04-05:
i've reinstalled sslstrip from the gui and installed it from tar.gz.
it's working now ) 
whatsapp keeps falling over sometimes.
nessus starts for a second and goes red -  in module details it's shown as
```
   Nessus
  version 1.1
   Nessus install
   Nessus disabled. | start

  Nessus is disabled.
```
tried reinstalling - no joy.

killbop, 2015-04-05:
did you had any weird problems on Kali or is it just me? 
recommended platform suggestions?

xtr4nge, 2015-04-05:
Hi killbop,
Looks like you have a permission problem or the installation is messed up.
I've installed FruityWifi several times on Kali Linux using apt-get and also from the GitHub repository and I didn't have any problem. You can debug it using the fruitywifi sudo log.
In Kali (apt-get installation) are in: `/var/log/fruitywifi/sudo.log`
GitHub installation (default path) are in: `/usr/share/fruitywifi/logs/sudo.log`

Regarding Nessus, FruityWifi module is only an interface, you need to download and install Nessus from tenable website.

regards,
xtr4nge

killbop, 2015-04-05:
ok, i went through the sudo.log and  couldn't see any events indicating problem
you're free to check for yourself
http://www.fileconvoy.com/dfl.php?id=g570b4aef5255cc599996464234b4ccdb9ced80c9b

xtr4nge, 2015-04-11:
Try this:
`chmod 755 /usr/share/fruitywifi/www/modules/sslstrip/includes/sslstrip-master/sslstrip.py`

Let me know.
regards,

closed 2015-05-30 by xtr4nge