xtr4nge/FruityWifi #109: Installation & Modules

pairughdocks, 2015-04-07:
I am using Kali Linux 64bit in a VMWare player (installed, not live). I have done a "git clone" of the repo and a "apt-get" to install FruityWifi and neither time does it work out 100% as instructed. I can finally get it to install and then when I get to the modules I can either not start anything or if a module does start it will instantly disable. I have an eth0 and a wlan0 adapter on my VM both are bridge directly to my physical network. Please advise of possible troubleshooting steps.

Thank you.

xtr4nge, 2015-04-07:
Hi pairughdocks,
About the interface, Not sure if I'm understanding your setup. On a VM machine, you need to have a wifi USB dongle as a physical device. This means that you need to see the dongle/wireless_card (from your VM) as wlan0. You can test the wireless_card putting it in monitor mode (from the VM) using for example `airmon-ng start wlan0`

About the problem of changing the config, it looks like a permission problem. Check if the file `/etc/sudoers.d/fruitywifi` has the permissions root:root

If `http://localhost:8000` and/or `https://localhost:8443` are giving you `.` , then, there is definitely a problem with the installation as both ports (8000|8443) points to `/usr/share/fruitywifi/www` and not to `/var/www`

regards,
xtr4nge

pairughdocks, 2015-04-07:
Thanks for the prompt reply xtr4nge,

My setup is:
Kali Linux x64 in VM
eth0 - physical - bridged
wlan0 - physical usb - bridged (Alfa with RTL8187 chipset)
mon0 - wlan0 monitor interface - successful - works with all aircrack products
Host OS - Windows 7 Pro x64

I will check my permissions file for sure, but I am running as root on Kali during the install.

When I perform the apt-get method, both ports 8000 and 8443 point to /usr/share/fruitywifi/www but if I just clone the git repo the install-FruityWifi.sh script configures the path as /var/www and using port 80/443. I also have to generate an SSL cert when using that script. It's not like any guide on the website or video I have seen in regards to the configuration of FruityWifi.

Thank you again,
pairughdocks.

xtr4nge, 2015-04-08:
Hi pairughdocks
If you install it using apt-get you don't need to install from GitHub.
If you install from GitHub it is not enough with cloning the repository, you need to download the content from GitHub and exec the installer `install-FruityWifi.sh` that you will find inside the downloaded zip file.

You will find port 80 and 443 open after using the GitHub script (installer) but it is not for running FruityWifi (only accessible on ports 8000 and 8443), they are needed by some modules that requires a webserver on port 80 and 443.

Note that the Alfa RTL8187 will not work with Hostapd as it is a Realtek chipset, so you can't use the modes: Hostapd, Karma and/or Mana. You can only use the [AP] mode Airmon-ng.

regards,
xtr4nge

closed 2015-04-25 by xtr4nge