xtr4nge/FruityWifi #111: Mana

lnkakalink, 2015-04-10:
handle_probe_req: send failed
 MANA - BROADCAST RESPONSE : FreeWifi:) (10) for STA 68:
 MANA - GENERATED SSID response for FreeWifi:) (len 10) :)
 MANA - FAILED SENDING PROBE RESP FOR SSID FreeWifi:) (10)
 MANA - BROADCAST RESPONSE : anton-home (10) for STA 68
 MANA - GENERATED SSID response for anton-home (len 10) :)
 MANA - FAILED SENDING PROBE RESP FOR SSID anton-home (10)
 MANA - BROADCAST RESPONSE : 17cab (5) for STA 68
 MANA - GENERATED SSID response for 17cab (len 5) :)
 MANA - FAILED SENDING PROBE RESP FOR SSID 17cab (5)
handle_probe_req: send failed
 MANA - BROADCAST RESPONSE : FreeWifi:) (10) for STA 68
 MANA - GENERATED SSID response for FreeWifi:) (len 10) :)
 MANA - FAILED SENDING PROBE RESP FOR SSID FreeWifi:) (10)
 MANA - BROADCAST RESPONSE : anton-home (10) for STA 68
 MANA - GENERATED SSID response for anton-home (len 10) :)
 MANA - FAILED SENDING PROBE RESP FOR SSID anton-home (10)
 MANA - BROADCAST RESPONSE : 17cab (5) for STA 68:
 MANA - GENERATED SSID response for 17cab (len 5) :)
 MANA - FAILED SENDING PROBE RESP FOR SSID 17cab (5)
handle_probe_req: send failed
 MANA - BROADCAST RESPONSE : FreeWifi:) (10) for STA 68:
 MANA - GENERATED SSID response for FreeWifi:) (len 10) :)
 MANA - FAILED SENDING PROBE RESP FOR SSID FreeWifi:) (10)
 MANA - BROADCAST RESPONSE : anton-home (10) for STA 68:
 MANA - GENERATED SSID response for anton-home (len 10) :)
 MANA - FAILED SENDING PROBE RESP FOR SSID anton-home (10)
 MANA - BROADCAST RESPONSE : 17cab (5) for STA 68:
 MANA - GENERATED SSID response for 17cab (len 5) :)
 MANA - FAILED SENDING PROBE RESP FOR SSID 17cab (5)
handle_probe_req: send failed

Jessexd, 2015-04-29:
consider compiling mana-toolkit and installing from source

xtr4nge, 2015-06-06:
Mana module has been updated.

regards,
xtr4nge

closed 2015-09-05 by xtr4nge