xtr4nge/FruityWifi #112: Can't get nessus module to enable

joeyj2468, 2015-04-15:
Hello,

I hope you are well.  I have been having issues with the modules, especally nessus.  When I try to start it, it still goes disabled.  Is something needed for nessus to work? if so, will that include the rest of the modules.  Nmap still gives me an error when I start it up.  Please advise.  Thank you.

[image]

xtr4nge, 2015-04-15:
Hi joeyj2468, I hope you are well.
You need to download and install Nessus from tenable website first. FruityWifi Nessus module is only an interface to interact with Nessus.

You need to have nmap installed and it cannot be used from the status/wsdl page, you can only use it from the Nmap module page.

regards,
xtr4nge

joeyj2468, 2015-04-15:
Ok.  Is there a nessus arm for raspberry pi? I have been trying to install it, but it appears there is no version.  I could be wrong and maybe looking in the wrong place.  Any leads to installing nessus on raspberry pi would be helpful.  Thanks.

xtr4nge, 2015-04-15:
Hi joeyj2468,
There is no Nessus ARM version yet.
You can setup a remote server from Nessus module, or you can install FruityWifi on another architecture.

regards,
xtr4nge

closed 2015-04-15 by xtr4nge