xtr4nge/FruityWifi #113: Can't get internet whilst FruityWifi is running [wlan: AR928X eth: BCM57780 OS: Kali]

Jessexd, 2015-04-26:
I have tried many combination and settings but I can't get internet I have read all of what is apparently is able to be called "wiki" and the "how tos" but I can't seem to get it running, my laptop even stops internet connection.

>gnome-network-manager and same reults with wicd

am I missing something?

Jessexd, 2015-04-29:
well, I got it "solved". For me, it did not work on it's own, neither git nor apt repo (and I will tell you why) some of fruitywifi modules and it's hostdap configuration where not working correctly no matter what, but it didnt really matter once it was running.

What I did was this:
downloaded Jessie debian (tested on purely sid and it also worked). 
Firstly, after installing Jessie ofc, I added all of  repos along with updated list of Jessie (sid works too) repos found here:

>http://docs.kali.org/general-use/kali-linux-sources-list-repositories
and here
>http://debgen.simplylinux.ch/

and configured /etc/apt/sources.list accordingly 

I blacklisted mana-toolkit because it caused problems and would not install, understandable because you arent running kali exactly. so you need to compile and install it from source.

>$ sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
>$ wget http://repo.kali.org/kali/pool/main/m/mana-toolkit/mana-toolkit_0~20140915-0kali1.debian.tar.gz
>$ sudo chmod +x  && ./kali-install.sh 
># Keep mana folder until end of tut
># Make your own judgement here, you could just make && make install but this script pulls in some dependencies needed for you 

>$ sudo apt-mark hold mana-toolkit

This will prevent conflict when installing fruitywifi

Now that that package is held back you can continue to install fruitywifi from official Kali Linux repo. 

>$ sudo apt-get install php5-fpm php5 nginx fruitywifi

You may see php5-fpm php5 nginx as unnecessary if installing fruitywifi with apt but it didnt install it on it's own.

Now to get a working hostdap + internet connection!

Go to path/to/mana/run-mana and run one of the scripts, start-nat-full.sh worked for me.
open up http://localhost:8000/page_config_adv.php and select hostadp-mana in drop down menu (only shows up with running above script)

You should now have a working AP with a working internet connection

xtr4nge, 2015-04-29:
Hi Jessexd,
Thanks a lot for the details provided.
I'm currently using `Wheezy` to develop FruityWifi so not sure why it is not working for you, but glad that you solved the problem.
I can see that they change something in Mana repository (Github). I will create a fork to have control on the updates.

Thanks and regards,
xtr4nge

Jessexd, 2015-04-29:
I suspect that it has to do with my hardware and Weezy being outdated. Jessie's new firmware, kernel, modules etc etc may be why it works for me. Best of luck.

xtr4nge, 2015-05-30:
Hi Jessexd,
Cool, glad that you solved the problem ;)

regards,
xtr4nge

closed 2015-05-30 by xtr4nge