xtr4nge/FruityWifi #114: PHP error issues

servarevitas3, 2015-04-27:
On the config page (or any page for that matter), when I select an interface, start wireless, etc, the change is not actually applied and nothing happens.

I checked error.log, and it is filled with errors like this: 

```
2015/04/27 01:32:57 [error] 4592#0: *181 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined index: iface in /usr/share/fruitywifi/www/scripts/config_iface.php on line 28
PHP message: PHP Notice:  Undefined index: io_in_iface in /usr/share/fruitywifi/www/scripts/config_iface.php on line 30
PHP message: PHP Notice:  Undefined index: io_in_iface_extra in /usr/share/fruitywifi/www/scripts/config_iface.php on line 31
PHP message: PHP Notice:  Undefined index: wifi_supplicant in /usr/share/fruitywifi/www/scripts/config_iface.php on line 32
```

I have reinstalled (from the kali repos) multiple times, as well as reinstalling php5-fpm.

Nothing seems to fix it.

Any ideas?

Thanks!

Jessexd, 2015-04-27:
I had this issue, it was fixed by resolving issues with dependancies relating to https://packages.debian.org/wheezy/amd64/php5-common
https://packages.debian.org/wheezy/amd64/php5-fpm

>$ apt-get purge php5-common php5-fpm
>$ apt-get clean && apt-get autoclean && apt-get update
>$ apt-get update
>$ dbpkg -i php5-common*.deb php5-fpm*.deb
And just in case
>$ apt-get --fix-missing

xtr4nge, 2015-04-29:
Hi Jessexd,
Thanks a lot for replying to this issue ;)

servarevitas3, please also check the permissions on `/etc/sudoers.d/fruitywifi` (needs to be `root:root`)

regards,
xtr4nge

servarevitas3, 2015-04-29:
Will do

Will report back ASAP

Jessexd, 2015-04-29:
>PHP message: PHP Notice:  Undefined index: io_in_iface in /usr/share/fruitywifi/www/scripts/config_iface.php on line 30
PHP message: PHP Notice:  Undefined index: io_in_iface_extra in /usr/share/fruitywifi/www/scripts/config_iface.php on line 31
PHP message: PHP Notice:  Undefined index: wifi_supplicant in /usr/share/fruitywifi/www/scripts/config_iface.php on line 32

This shows:
>regex_standard($_POST["io_in_iface"], "../msg.php", $regex_extra);
regex_standard($_POST["io_in_iface_extra"], "../msg.php", $regex_extra);
regex_standard($_POST["wifi_supplicant"], "../msg.php", $regex_extra);

wpa_supplicant is installed right? do you have the correct firmware for your wireless card installed as well as the interface being up? e.g. nonfree firmware.

show output of
$ ifconfig -a
$ lspci | grep -i wireless

servarevitas3, 2015-04-30:
Reinstalling both php5 debs (the armhf versions) did not fix the issue.
I also ran chown -R root:root /etc/sudoers.d/fruitywifi
Wifi adapters are working fine (external adapter works great using wifite, etc.)
This is on the kali chroot on nethunter. Is there a reason this would not run on nethunter?

Jessexd, 2015-04-30:
post the output of 
service php5 status
service php5-fpm status
service wpa_supplicant status
service fruitywifi status
service apach2 status

servarevitas3, 2015-04-30:
root@kali:~# service php5 status
php5: unrecognized service
root@kali:~# service php5-fpm status
[ ok ] php5-fpm is running.
root@kali:~# service wpa_supplicant status
wpa_supplicant: unrecognized service
root@kali:~# service fruitywifi status
[ ok ] fruitywifi is running.
root@kali:~# service nginx status
[FAIL] nginx is not running ... failed!
root@kali:~# service apache2 status
Apache2 is NOT running.
root@kali:~# 

I get the exact same result on my laptop, where fruitywifi is working and has been working fine.

Jessexd, 2015-05-01:
apt-get install apache2 
apt-get install php5-common
apt-get install nginx-full
fuser --kill 80/tcp && fuser --kill 443/tcp
service nginx restart && service nginx force-reload && service apache2 restart &&  service apache2 force-reload && service fruitywifi restart

servarevitas3, 2015-05-09:
I don't think it's an adapter issue, as the PHP errors I'm getting are not wireless-specific. 
Here is a link to the entire error.log:

https://gist.github.com/servarevitas3/9d89f4f1f75cd19c6716#file-error-log

Jessexd, 2015-05-10:
It may be a firewall issue it seems.
Check your router/modem settings

servarevitas3, 2015-05-10:
It's not that. I'm accessing it through the browser on the device itself.

Jessexd, 2015-05-10:
reread, I updated the message a little bit after I made it.

servarevitas3, 2015-05-10:
Not a port in use issue
Ran those commands anyway; no go.
I can load the interface and log in; I just can't alter the config (the page reloads without any changes values persisting) or start any modules.

Jessexd, 2015-05-10:
Can you uninstall it all and run the install script from, then post here about any errors you might have
https://github.com/xtr4nge/FruityWifi

The funky thing with it is that it will say its installed without any problem, but you can still scroll up in the terminal and see like a million errors like installing dependencies etc. So go ahead and install from the script included and see if you can find anything that is screwing up during install.

closed 2015-06-27 by xtr4nge