xtr4nge/FruityWifi #117: Ideas for moduler

lillstrumpa, 2015-05-06:
- Driftnet modul
- wlan txpower booster modul
- wiki module ( for fruitywifi where you can learn how to use use the module right, 
  Step by step what moduler you need to enable to get krama / mana to work , whats the differense of the modules

Maybe a chart where you can see what wificard is supported and so on ...

xtr4nge, 2015-05-30:
Hi lillstrumpa,
I hope you are well. Thanks a lot for your feedback and ideas.
I have Driftnet module pending, I will try to publish it as soon as possible.
Yes, I think we need a place to share information (wiki or forum). I'm working on that =)

regards,
xtr4nge

lillstrumpa, 2015-05-30:
I did just bought the wifipineapple mark V  but I like the wififruity better , ( I have it on my rpi B+ ) running om kali linux .. 

a tx booster would be nice ,

A few new Ideas is to implant :

wifite
wash with pixiedudt support
https://github.com/t6x/reaver-wps-fork-t6x
pixedust
is it possible to make an online DB with WPA password lists that wififruity can work against
so you dont need to have a big password list installed.