xtr4nge/FruityWifi #121: Supplicant, nmcli cant connect to wifi

Newpid0, 2015-06-03:
I am trying to get FruityWifi to connect to my wireless. I have tried using the modules supplicant/nmcli they both say installed. however when i click start nothing happens. I am sure I am missing something simple but I can't seem to see what it is. I have even tried adding various wireless usb. If I use my alfa and Ethernet things work great. But I would like to add a second usb wifi adapter and use that to connect to my jetpack hotspot. I have also tried various wpa modes, ssid names and even tried various passwords. hoping to eliminate any weird character issues or something. I even tried to connect with my home wifi in the off chance it was the jetpack. still no luck. 

FruityWifi: V2.1.2
Supplicant:v1.2
nmcli: v1.3
Raspberry Pi2 

any help would be great

Newpid0, 2015-06-03:
took some tinkering and Im not even sure what fixed it but i got it working

closed 2015-06-03 by Newpid0

xtr4nge, 2015-06-04:
Hi Newpid0,
I'm glad that you solved the problem.
Thanks a lot for let me know.

regards,
xtr4nge