xtr4nge/FruityWifi #122: Modules issues

Newpid0, 2015-06-04:
As I was trying add some modules I noticed many of the modules, such as Mana, Karma and even autostart reference this exact line when  in the main index.php enabling "<a href=\"includes/module_action.php?service=responder&actio" seems like its trying to enable responder which doesn't make sense to me. Like why is this line in autostart. https://github.com/xtr4nge/module_autostart/blob/master/index.php#L93
something seems off to me

Newpid0, 2015-06-06:
If you can confirm if I am correct in my thinking here I can make some pull requests to help out

xtr4nge, 2015-06-06:
Hi Newpid0,
`module_action.php` does not need to know the name of the module anymore. It only checks if `service` variable is empty or not =)

regards,
xtr4nge

closed 2015-06-27 by xtr4nge