xtr4nge/FruityWifi #123: mana and karma

Newpid0, 2015-06-06:
If anyone knows a good spot to read up on actually getting karma or mana running with fruitywifi the newest version. I would love to know where. fruity seems great, but i just cant get either running. and running the commands to enable mana or karma manually generally give me a "unknown command error"
Any thoughts would be great.

xtr4nge, 2015-06-06:
Hi Newpid0,
Mana module has been updated. It was a problem with the installer as the original repository was moved.
Please note that you need to change the setup in the config page (`Hostapd-Mana` or `Hostapd-Karma`) before to start wireless and use mana or karma.

regards,
xtr4nge

Newpid0, 2015-06-06:
mana installed great. Thanks for the update. The only thing i noticed is I don't have the options for Hostapd-Mana or Hostapd-Karma on the config page. Is there something I am missing. Btw nice work on this, it really is a great project.

xtr4nge, 2015-06-06:
Hi Newpid0,
You need to change the [AP] option to `Hostapd-Mana`.
If the option is not there, the module is not fully installed. Enter the module and check if all the dependencies are installed.

regards,
xtr4nge

Newpid0, 2015-06-06:
I will check it out. Thanks

Newpid0, 2015-06-06:
Found part of the issue the config page calls 
/usr/share/FruityWifi/www/modules/mana/includes/hostapd
however fruity installs to 
/usr/share/fruitywifi/www/modules/mana/includes/hostapd

Case sensitive path issue with FruityWifi vs fruitywifi

https://github.com/xtr4nge/FruityWifi/blob/master/FruityWifi/www/page_config_adv.php#L235
and 
https://github.com/xtr4nge/FruityWifi/blob/cc3156cdef4c3c7ada95582687fcbef4e47a506c/FruityWifi/www/page_config_adv.php#L238

xtr4nge, 2015-06-24:
Hi Newpid0,
Cool, thanks for the info. I will fix the code ;)

regards,

closed 2015-06-24 by xtr4nge