xtr4nge/FruityWifi #124: Pivoting via Meterpreter module

M4dMax, 2015-06-06:
Hi xtr4nge, i'm a novice in infosec and im very glad that you guys wrote so amazing tool like fruitywifi.
My setup is as follows, i installed kali arm version on a raspberry pi model b, than i did an apt-get install fruitywifi after an decent update and upgrade. Fruitywifi start good . i have an alfa awus036-nh on my pi. Only hostap-mana seems to give internet to my clients that are connected on my fruitywifi
I was trying to make fruitywifi work together with armitage by pointing the meterpreter module to a listener that i make with armitage .And as soon as i configure and start the meterpreter module i get a shell to my pi, but i have problem to pivot to my connected clients.
My main target is to get an meterpreter session with every client the may connect to my fruitywifi. How can i achieve that???

Newpid0, 2015-06-06:
How are you getting the shell on the clients connected to your fruitywifi? What module are you using to get the shell?

M4dMax, 2015-06-06:
No, im not gettting a shell to my clients. im only getting a shell to my rasp pi(kali arm+ fruitywifi). what i'm doing is , pointing the meterpreter module to my listener and than i get a shell to the pi. i even logged in via ssh (using armitage)to my pi,then i run ssh to meterpreter postmodule to upgrade to a meterpreter, but i can pivot through it. I want pwn my connected clients

Newpid0, 2015-06-06:
To pwn your clients you need a method to do so. Such as an exploit or a
phish or serve up an exe on the fruity for them to run. I may be missing
what you are saying but I think I get what you are saying

M4dMax, 2015-06-06:
Allright. Thank for the reply. You mean that i have to run beefstrike in armitage to hook my client's browsers and than i can pivot from my pi to the clients?

Newpid0, 2015-06-06:
Yes that or something like that. Just them joining your fruitywifi wont
give you shell.

M4dMax, 2015-06-06:
Ok, now i get it. im right now doing some test in my lab. my goal is to have armitage running together with beefstrike so i can hook connected clients and than some how inject the meterpreter in each one of them. I'll appreciate very much if you could guide me or give me some good suggestions to configure all those tools to work together (fruitywifi+meterpreter module+armitage+beefstrike)

Newpid0, 2015-06-06:
I have never used armatage or beefstrike I'd suggest google

M4dMax, 2015-06-06:
Thanks for your time my friend

xtr4nge, 2015-08-30:
Please re-open this thread in the forum if required: http://fruitywifi.boards.net/

regards,

closed 2015-08-30 by xtr4nge