xtr4nge/FruityWifi #125: use updated sslstrip version with dns2proxy

twayz, 2015-06-06:
Hi,

I was just wondering if you could use the updated sslstrip module along with dns2proxy (this combination partially removes hsts headers and results in more successfully stripping ssl than the current version)? The updated module requires dns2proxy - both are found here - sensepost's repo https://github.com/sensepost/mana/tree/master/sslstrip-hsts

thank you so much for this!

xtr4nge, 2015-06-07:
Hi twayz,
Thanks, I will take a look ;)

regards,
xtr4nge

twayz, 2015-06-07:
Sounds great! Also, I found MITMf to work best at stripping ssl. There was discussion on their repo about potentially integrating MITMf with fruitywifi. Here is the link to that discussion: https://github.com/byt3bl33d3r/MITMf/issues/31
Hope that helps,
Twayz

xtr4nge, 2015-06-07:
Hi twayz,
Thanks again for the info. I'm going to take a look and create some new modules for FruityWifi ;)
Very helpful information.

regards,
xtr4nge

xtr4nge, 2015-06-24:
Hi twayz,
The module MITMf has been released and it works really well ;)
Thanks a lot again for the info.

regards,
xtr4nge

closed 2015-06-24 by xtr4nge