xtr4nge/FruityWifi #130: eth0 Interface disapper ?

j0kara, 2015-06-09:
Hi, today i started playing with furitywifi and it looks really amazing, but i'm having a few problems with it. My problem is that my eth0 interface is dissaper or get disabled, but i don't know how to get it back.

When i type 'ifconfig' i don't see it, but when i type 'iwconfig' i can see - 'eth0      no wireless extensions.'

Any idea or is there any command how to get my interface back, reboot don't help out :/

(I tried on Raspbian OS, on RPI and also on kali linux at regular laptop, issue happening at both machines)

demogorgonz, 2015-06-10:
Try : 
ifup eth0
or
ifconfig eth0 up

Replace eth0 with your interface name...

j0kara, 2015-06-10:
I already tried 'ifconfig eth0 up' and after that it shows up my eth0 interface but i have no internet. I also tried to reinstall Raspbian and then it work again but problem happend again after a hour or so.

xtr4nge, 2015-08-30:
Please re-open this thread in the forum if required: http://fruitywifi.boards.net/

regards,

closed 2015-08-30 by xtr4nge