xtr4nge/FruityWifi #133: Unable to view Kismet logs

MarquisT, 2015-06-22:
When I click the export of a kismet log, I don't see anything. This happens in all browsers. I am using RPI. Suggestions?

xtr4nge, 2015-06-23:
Hi MarquisT,
Please check that kismet it is fully installed. When you download a module in FruityWifi, this action does not install the module. To complete the installation you need to access the module and verify that all the dependencies are installed. For example, after download Kismet and entering the module, you will see `install` in red (if everything is installed you will see `installed` in green). Press install and wait for the dependencies to be downloaded/compiled/installed.

The export button does not download the content. After press `export` you will see a `download` button to get the content.

regards,
xtr4nge

MarquisT, 2015-06-27:
Hi.. 
I'm able to open the raw Kismet logs. When I click that export button I don't get the download button.

xtr4nge, 2015-06-27:
Hi MarquisT,
Please notice that you get content to be downloaded only when you are using Kismet + GPS. The file created is a KML, and can be displayed, for example, using Google Earth.

I know that this information it is not in any documentation. I will try to add documentation per module as soon as possible.

regards,
xtr4nge

xtr4nge, 2015-08-30:
Please re-open this thread in the forum if required: http://fruitywifi.boards.net/

regards,

closed 2015-08-30 by xtr4nge