xtr4nge/FruityWifi #135: DNS Issues

AnguisCaptor, 2015-06-27:
I'm not sure if anyone is having problems with Fruity and DNS but I noticed that when using it on the raspberry pi 2, there is no internet really as the DNS is kept locked at 127.0.0.1

Is there a easy way to change it from 127.0.0.1 to let's say 8.8.8.8 ?

xtr4nge, 2015-06-27:
Hi AnguisCaptor,

When you enabled `Wireless` from the status page, FruityWifi modifies the file `/etc/resolv.conf` adding 2 entries, example:

`nameserver 10.0.0.1`
`nameserver 8.8.8.8`

Note that the IP `10.0.0.1` will be changed by the IP that you put in the config page. (by default 10.0.0.1)

regards,
xtr4nge

AnguisCaptor, 2015-06-27:
Interesting... I guess I had to reboot my raspberry
When I take a look now in the resolv.conf I see my gateway IP but no 8.8.8.8

xtr4nge, 2015-06-27:
The setup of `/etc/resolv.conf` changes every time that you obtain a new IP/config (DHCP)

regards ;)

AnguisCaptor, 2015-06-28:
Today again... DNS went again to 127.0.0.1, which means no internet really, is there a way to lock it to 8.8.8.8 ?

Jessexd, 2015-06-28:
You can, setup your fruity wifi ip settings statically, also setting them in the resolv.conf

then after settings are changed:
# chattr +i /etc/resolv.conf

You may see error warnings from some of the fruity  wifi modules but it will still go through as long as it's all correctly set.

AnguisCaptor, 2015-07-16:
Kindof a late response, but I think I've solved it now using the command Jessexd showed
sudo echo "nameserver 208.67.222.222" > /etc/resolv.conf && chattr +i /etc/resolv.conf

closed 2015-07-16 by AnguisCaptor

xtr4nge, 2015-07-16:
cool, thanks for the update ;)

regards,