xtr4nge/FruityWifi #136: FruityProxy missing libmproxy

AnguisCaptor, 2015-06-27:
I was about to try out FruityProxy till I noticed it tried to start but failed, so I ran fruityproxy.sh manually and quickly came up what was missing

Traceback (most recent call last):
  File "fruityproxy.py", line 20, in <module>
    from libmproxy import controller, proxy
ImportError: No module named libmproxy

xtr4nge, 2015-06-27:
Hi AnguisCaptor,
Thanks a lot for reporting this.
Please run the script `install.sh` manually: 
Path: `FruityProxy-master/install.sh`
Full path: `/usr/share/fruitywifi/www/modules/fruityproxy/includes/FruityProxy-master/install.sh` 

looks like MITMproxy is missing.

please let me know if that solved the problem ;)

regards,
xtr4nge

AnguisCaptor, 2015-06-27:
I tried that earlier but apparently Raspbian didn't have everything up-to-date, so I tried
pip install pyopenssl --update

But this failed at the very end so I tried it again untill this showed up "from pkg_resources import load_entry_point"
So I fixed that error by using "wget https://bitbucket.org/pypa/setuptools/raw/bootstrap/ez_setup.py -O - | python"

Now I tried "pip install mitmproxy" and it's currently installing for quite a while now... will let you know when it's done ;)

xtr4nge, 2015-06-27:
Awesome ;)
Thanks a lot for the info, I didn't tried the installation of FruityProxy on R-Pi yet.
Let me know, and if you can, do a writeup of the steps as this can help a lot of people ;)

regards,

AnguisCaptor, 2015-06-27:
Oh, I currently got it working! this are the exact steps I took to get it working
Some info, I ran Raspbian with the lastest updates on the PI 2
when doing "pip install mitmproxy", I was getting a error showing "Requirement.parse('pyOpenSSL>=0.15.1')"
So then I did the obvious thing of doing "pip install pyopenssl --upgrade"
But at the end it gave some weird error so I ran it once more but then "from pkg_resources import load_entry_point" showed up
So I fixed that by doing "wget https://bitbucket.org/pypa/setuptools/raw/bootstrap/ez_setup.py -O - | python"
Now I tried one more time "pip install mitmproxy", after ~10 minutes it was installed!
So I tried the command "mitmproxy" or your fruityproxy.sh but that was something along the lines of "missing flask"
So then I ran "pip install flask", and now I tried fruityproxy.sh or mitmproxy, and it worked!

So in the end #tl;dr
# pip install mitmproxy
# pip install pyopenssl --upgrade
# wget https://bitbucket.org/pypa/setuptools/raw/bootstrap/ez_setup.py -O - | python
# pip install mitmproxy
# pip install flask
Now run Fuity Proxy :)

xtr4nge, 2015-06-27:
Cool, thanks for the info!
I will add Flask to the installer and the info to the FruityProxy wiki.
If you want to make some plugins, and play with the API, just check the wiki.

Give me some feedback after test FruityProxy ;)

regards,

xtr4nge, 2015-07-16:
Hi AnguisCaptor,
Installer for FruityProxy has been updated. Thanks ;)

regards,

closed 2015-07-16 by xtr4nge