xtr4nge/FruityWifi #137: Can't do anything. Loads and nothing happens.

Devlon, 2015-06-29:
Installed Fruitywifi on rip 2 and latest Raspbian.
After following instructions to install install-FruityWifi.sh and install-modules.py all seemed to go well.
logging in and all looks fine.

Problem is that I can't do a single thing accept reveal public ip. 
When I click to enable any module it loads, shows the loading fruit page, goes back,  then nothing. Can't even remove installed modules.

Any suggestions?

xtr4nge, 2015-06-29:
Hi Devlon,
Did you check the config page? (you need to setup the interfaces first)
If your wireless card is not supported by Hostapd (such us Realtek chipsets), you can use `Airmon-ng` [AP] mode only. (Hostapd, Hostapd-Karma and Hostapd-Mana will not work)
Also check that the default Raspian setup it is not conflicting with the Interfaces.

regards,

Devlon, 2015-06-29:
Yes same thing applies in the config page. Whatever I choose it returns back to default. The only thing that actually changes is public IP ;)

xtr4nge, 2015-06-29:
Looks like a permission problem.

There are a couple of things that you need to verify:
- Check that the user fruitywifi exists.
- Check that the file `/etc/sudoers.d/fruitywifi` exists. (root:root permission)
- The ports that you need to use to access FruityWifi are 8000 and/or 8443
- FruityWifi path needs to be `/usr/share/fruitywifi/`

You can re run the installation script, and check the output to see if something is crashing 
Note: the script does not check for errors, but you will see the errors on the screen.

regards,

Devlon, 2015-06-29:
Thnx!
Will check and repory tomorrow..

Devlon, 2015-06-29:
Check that the user fruitywifi exists. 
fruitywifi:x:108:111:FruityWifi,,,:/var/run/fruitywifi:/bin/false

Check that the file /etc/sudoers.d/fruitywifi exists. (root:root permission)
fruitywifi      ALL=(root:root) NOPASSWD:ALL    (Changed from ALL:ALL to root:root)
Defaults:fruitywifi loglinelen = 0
Defaults:fruitywifi logfile = /usr/share/fruitywifi/logs/sudo.log
Defaults:fruitywifi log_year
Defaults:fruitywifi log_host
Defaults:fruitywifi syslog=auth

The ports that you need to use to access FruityWifi are 8000 and/or 8443
Only able to connect :80000

FruityWifi path needs to be /usr/share/fruitywifi/
It's there.

Can't sleep..

xtr4nge, 2015-06-29:
you can debug it tailing FruityWifi sudo log file:
`/usr/share/fruitywifi/logs/sudo.log`

Every time that you perform an action, you will see some entries in the log file as all commands are executed using sudo (fruitywifi user)

Also check the Nginx and php5-fpm setup as you have a problem there if you cannot connect to port 8443.

regard,

Devlon, 2015-06-29:
cat: /usr/share/fruitywifi/logs/sudo.log: No such file or directory

pi@raspberrypi ~/FruityWifi-master $ sudo ./install-FruityWifi.sh 
sudo: /etc/sudoers.d/fruitywifi is owned by uid 1000, should be 0
mkdir: cannot create directory `tmp-install': File exists
Hit http://mirrordirector.raspbian.org wheezy Release.gpg
Hit http://raspberrypi.collabora.com wheezy Release.gpg                                         
Hit http://archive.raspberrypi.org wheezy Release.gpg                                           
Hit http://raspberrypi.collabora.com wheezy Release                                             
Hit http://mirrordirector.raspbian.org wheezy Release                                           
Hit http://archive.raspberrypi.org wheezy Release                    
Hit http://apt.adafruit.com wheezy Release.gpg                        
Hit http://raspberrypi.collabora.com wheezy/rpi armhf Packages        
Hit http://mirrordirector.raspbian.org wheezy/main armhf Packages    
Hit http://archive.raspberrypi.org wheezy/main armhf Packages                                                 
Hit http://apt.adafruit.com wheezy Release                                                                     
Hit http://mirrordirector.raspbian.org wheezy/contrib armhf Packages                                           
Hit http://mirrordirector.raspbian.org wheezy/non-free armhf Packages                    
Hit http://mirrordirector.raspbian.org wheezy/rpi armhf Packages                                               
Hit http://apt.adafruit.com wheezy/main armhf Packages                                                         
Ign http://raspberrypi.collabora.com wheezy/rpi Translation-en_GB                                              
Ign http://raspberrypi.collabora.com wheezy/rpi Translation-en                           
Ign http://archive.raspberrypi.org wheezy/main Translation-en_GB                         
Ign http://archive.raspberrypi.org wheezy/main Translation-en                            
Ign http://apt.adafruit.com wheezy/main Translation-en_GB          
Ign http://apt.adafruit.com wheezy/main Translation-en             
Ign http://mirrordirector.raspbian.org wheezy/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org wheezy/contrib Translation-en
Ign http://mirrordirector.raspbian.org wheezy/main Translation-en_GB
Ign http://mirrordirector.raspbian.org wheezy/main Translation-en
Ign http://mirrordirector.raspbian.org wheezy/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org wheezy/non-free Translation-en
Ign http://mirrordirector.raspbian.org wheezy/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org wheezy/rpi Translation-en
Reading package lists... Done                                                                                                                
--------------------------------
Creates user fruitywifi
--------------------------------
[fruitywifi user has been created]

Reading package lists... Done
Building dependency tree       
Reading state information... Done
autoconf is already the newest version.
automake is already the newest version.
build-essential is already the newest version.
curl is already the newest version.
dos2unix is already the newest version.
gettext is already the newest version.
intltool is already the newest version.
php5-cli is already the newest version.
php5-curl is already the newest version.
sudo is already the newest version.
unzip is already the newest version.
uuid is already the newest version.
uuid-dev is already the newest version.
make is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--------------------------------
gcc 4.7 already installed
--------------------------------

--------------------------------
dnsmasq already installed
--------------------------------

--------------------------------
hostapd already installed
--------------------------------

[... 60 lines cut ...]

--------------------------------
Installing danger
--------------------------------

--------------------------------
Start Services
--------------------------------
update-rc.d: using dependency based boot sequencing
update-rc.d: warning: default stop runlevel arguments (0 1 6) do not match ssh Default-Stop values (none)
update-rc.d: using dependency based boot sequencing
update-rc.d: using dependency based boot sequencing
update-rc.d: using dependency based boot sequencing
update-rc.d: warning: default stop runlevel arguments (0 1 6) do not match ntp Default-Stop values (none)
Restarting nginx: nginx.
[ ok ] Restarting PHP5 FastCGI Process Manager: php5-fpm.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'ifplugd' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

GitHub: https://github.com/xtr4nge/FruityWifi
Twitter: @xtr4nge, @FruityWifi
ENJOY!

xtr4nge, 2015-06-29:
`sudo: /etc/sudoers.d/fruitywifi is owned by uid 1000, should be 0`

** Permissions on file `/etc/sudoers.d/fruitywifi` needs to be `root:root` (owner and group: root)

regards,

Devlon, 2015-06-29:
Sorry my bad, forgot about chown. 
Changed to root:root

pi@raspberrypi /etc/sudoers.d $ ls -al
total 16
drwxr-xr-x   2 root root 4096 Jun 29 18:00 .
drwxr-xr-x 116 root root 4096 Jun 29 19:58 ..
-rw-r--r--   1 root root  229 Jun 26 13:21 fruitywifi
-r--r-----   1 root root  958 Feb 22 09:38 README

Root:root did the trick! I can turn wifi on and off.
Must reinstall all modules now. 

Thank you so much for quick reply and keep up the greate work!

Best regards

xtr4nge, 2015-06-29:
No prob at all ;)

Thanks and regards,

closed 2015-06-29 by xtr4nge