xtr4nge/FruityWifi #138: PHP Header and Session errors

decapyre, 2015-07-14:
A lot of this happening because menu.php contains header() calls to remove caching. You can't have any whitespace or even html outputted before you call the header/session method.

Warning: session_start(): Cannot send session cache limiter - headers already sent (output started at /usr/share/fruitywifi/www/page_modules.php:49) in /usr/share/fruitywifi/www/login_check.php on line 20

Warning: Cannot modify header information - headers already sent by (output started at /usr/share/fruitywifi/www/modules/supplicant/index.php:42) in /usr/share/fruitywifi/www/modules/menu.php on line 24

Warning: Cannot modify header information - headers already sent by (output started at /usr/share/fruitywifi/www/modules/supplicant/index.php:42) in /usr/share/fruitywifi/www/modules/menu.php on line 25

Warning: Cannot modify header information - headers already sent by (output started at /usr/share/fruitywifi/www/modules/supplicant/index.php:42) in /usr/share/fruitywifi/www/modules/menu.php on line 26

Warning: Cannot modify header information - headers already sent by (output started at /usr/share/fruitywifi/www/modules/supplicant/index.php:42) in /usr/share/fruitywifi/www/modules/menu.php on line 27

Warning: Cannot modify header information - headers already sent by (output started at /usr/share/fruitywifi/www/modules/supplicant/index.php:42) in /usr/share/fruitywifi/www/modules/menu.php on line 28

xtr4nge, 2015-07-14:
Hi decapyre,
The headers will be removed from `menu.php` in the next release. 
What are the problems that you are having with modules? 

regards,
xtr4nge

decapyre, 2015-07-14:
None of the modules stay started. They start/stop after a second. I have a wireless RT5370 chipset and ethernet hooked up.

xtr4nge, 2015-07-14:
I know that RT5370 works very well with Raspbian, but I didn't tested it on Kali (ARM)

Please try/check the following options:

- Change /etc/network/interfaces: (to avoid interfaces conflict with FruityWifi)

`auto lo`
`iface lo inet loopback`
`allow-hotplug eth0`
`iface eth0 inet dhcp`
`#allow-hotplug wlan0`
`#iface wlan0 inet manual`
`#wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf`
`#iface default inet dhcp`

- Try [AP] mode Airmon-ng (karma and mana will not work with this mode)

- check the FAQ: https://github.com/xtr4nge/FruityWifi/wiki/FAQ

Let me know
regards,

decapyre, 2015-07-16:
- Fresh Raspbian install, upgraded and configured.
- Download master.zip
- unzip and run install script
- /etc/network/interfaces looks good.
- Start Wireless, it starts for a second and then stops.

Using RT5370 Dongle.

xtr4nge, 2015-07-16:
Same behaviour with Airmon-ng mode?

Check the new version (v2.2) from the DEV tree: https://github.com/xtr4nge/FruityWifi/archive/develop.zip

I will do some tests asap with the R-PI and the RT5370

regards,

decapyre, 2015-07-16:
Going to try the develop branch as you mentioned. This issue is now unreleated to the `PHP Header and Session warnings` so I'm going to close this

closed 2015-07-16 by decapyre

xtr4nge, 2015-07-17:
Hi decapyre 
I just tested R-PI2 + RT5370 and it works great.

lsusb: `:5370 Ralink Technology, Corp. RT5370 Wireless Adapter`

I just found a problem on R-PI with `resolveconf` that keeps reseting the file `resolv.conf` but It was resolved on the dev branch.

also check that you have commented the following lines on `/etc/network/interfaces` (at least wlan0 setup):

#auto wlan0
#allow-hotplug wlan0
#iface wlan0 inet manual
#wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

#auto wlan1
#allow-hotplug wlan1
#iface wlan1 inet manual
#wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

Please try again.
regards,

decapyre, 2015-07-17:
Yep, I confirm that the resolveconf was getting reset, I fixed it myself, but nice to see its in the dev branch!

RPI2 + RT5370 is working, thanks!

xtr4nge, 2015-07-17:
No prob ;)

I just added: 
`chattr +i /etc/resolv.conf` on start
and
`chattr -i /etc/resolv.conf` on stop

Please post any bug or suggestion, and I will take a look.

Thanks and regards,