xtr4nge/FruityWifi #140: Version 2.2 Issues

AnguisCaptor, 2015-07-17:
So far, goodjob at version 2.2, but I found a few issues so far

1. The "FruityWifi" logo/button at the top left only goes to the current module/page
2. There is a issue with karma or mana, or atleast there is with enabling Wireless, I have Wireless on at autostart but when I look at the main page, Wireless is not on at startup, when I look at my dongle, it's showing no light indication that it's turned on or active when I re-plug my dongle, I see light again and when I enable Wireless, it's off, so it looks like the Wireless option is turning the dongle off appearently lol
Using Hostapd instead of Hostapd-Mana/Karma seems to fix it, but mana does work on my dongle

3. Not sure if this is normal, but when FruityProxy is enabled, internet goes abnormal slow compared to when it's not enabled (Slow for stations connected to the AP only)

4.Old bug, not the fault of 2.2 but I'll put it in here anyway, the meterpreter module can only listen at 1 port default is 6666, but when you look at FruityProxy, BDFactory etc... you can go upto like 2-5 ports, or I'm using the module wrong haha

5. Sometimes when I disable FruityProxy, there is no internet anymore

xtr4nge, 2015-07-17:
Hi AnguisCaptor,
Cool, thanks a lot for reporting these issues, I will take a look ;)

regards,

xtr4nge, 2015-07-20:
Hi AnguisCaptor,

1- Logo/Button now goes to www.fruitywifi.com ;)

2- I couldn't check AutoStart module yet, but I fixed and tuned a lot of issues on Hostapd, Hostapd-Mana, Hostapd-Karma and Airmon-ng.

3- I updated FruityProxy and FruityProxy module. (please let me know if speed looks better now)
Note: This may depends/vary on how many FruityProxy plugins are ON at the same time and which ones, as the web traffic is being manipulated by each plugin.

4- I couldn't check Meterpreter module yet, but I will take a look as soon as I can.

Sub1- This cold be because some IPTABLES rule that is not removed, I will take a look.

Thanks ;)

AnguisCaptor, 2015-07-22:
1. Fixed
2. Autostart seems to work for other modules, but it's not starting the AP module / Karma
3. FruityProxy is really slow, having it enabled the website takes 27 seconds to load, having it disabled it takes 15 seconds, and this is without any modules enabled, I must say that 15 seconds is already slow but that might be the network where I'm at
4. cool :)

I noticed as well that the start/stop of AP is fixed as well that internet will stay, thanks ;)

closed 2016-09-05 by xtr4nge