xtr4nge/FruityWifi #143: Modules not starting in v2.2

DavidX001, 2015-07-20:
Hi,
I have "Alfa wifi", all module changes to enabled for a second, then changes to disabled automatically except wireless module.
help me :)

xtr4nge, 2015-07-20:
Hi DavidX001,
What chipset are you using (`lsusb`)? What OS/Linux are you using? Hardware?

Some chipsets are not compatible with the default setup that is [AP] Hostapd. Try using Airmon-ng if that is the case. 

please check the FAQ for common issues: https://github.com/xtr4nge/FruityWifi/wiki/FAQ

Please let me know.
regards,

DavidX001, 2015-07-20:
RT2870/RT3070 / raspberry pi 2 / Raspbian.
Airmon-ng dont work :)

Merci  :)

xtr4nge, 2015-07-20:
There are some reports confirming that it is not supported by Hostapd:

`https://github.com/xtr4nge/FruityWifi/issues/81` https://github.com/xtr4nge/FruityWifi/issues/81
`https://github.com/xtr4nge/FruityWifi/issues/61` https://github.com/xtr4nge/FruityWifi/issues/61

I don't have that chipset, so I can't confirm if works or not with Airmon-ng. 
Airmon-ng is an option that we added to support more chipsets, but it is not the recommended option, as some modules are not supported such us Karma either Mana.

to confirm if your card works with Airmon-ng try this command:

`airmon-ng start wlan0` (or the name of your wlan interface)
You need to see a new interface now, called: `mon0`

Try some checks on the new interface (`mon0`)
`airodump-ng mon0`
If you start seeing traffic, we can work out something with that card, if not, it is not supported either by `airmon-ng`

regards,

closed 2015-08-08 by xtr4nge