xtr4nge/FruityWifi #145: Modules (mana, karma, mitmf, etc.) not starting properly

LucaBongiorni, 2015-07-21:
My environment: Rpi2 + AR9271 + PwnPi 3.0

In many cases the modules are not starting properly or starting at all.
E.g. Sometimes If I set Hostapd- Mana and then I try to activate MANA modules, it will not get activated. The same behaviour with KARMA. 
Moreover I am unable to start MITMf or FruitiProxy with both hostpad-MANA / hostapd-KARMA.

Is there a possibility to enable some sort of DEBUG MODE where I can see all executed calls/cmds from the php GUI in one aggregated logfile?

P.S. As far as I understood some modules cannot be run at the same time (mutual exclusion). Would not be better to create a GUI which would specify in a more user-friendly way... what can be run at together and which not? [I know it might sounds dumb, but it will save time for newcomers and will avoid confusion/mistake during an ongoing pentest].

BR,
Luca

xtr4nge, 2015-07-21:
Hi LucaBongiorni ,
`MITMf` and `FruityProxy` both requires `BDFproxy` module fully installed first. It is not required to have BDFproxy module running, but needs to be installed.

It is true that will help a lot to have more information about what modules required what other module. I'm working on that to make a more friendly interface.

I will try to have the forum ready as soon as possible as that will help people to start helping each other as well.

Please let me know if BDFproxy module solved the problem.

regards,

LucaBongiorni, 2015-07-21:
I knew I was missing something.
I have install fresh new FW from github and ran python install-modules.py
Therefore I should have all modules already configured.
Right now I am digging to figure out why some modules are not running properly.

About the forum... it would be a great idea! ;]

P.S. FYI I just ran ./usr/share/FruityWifi/www/modules/bdfproxy/includes/bdf-proxy/test-script.sh and looks working properly.

xtr4nge, 2015-07-21:
Cool, good to know that the problem was solved ;)

Yes, the forum will help a lot!

If you have any other query, please let me know.
regards,

LucaBongiorni, 2015-07-21:
Actually, what I meant, is that I have already installed all modules in advance. So problem is still there.
To be precise, now, I have running hostapd-mana + mana module. But it fails to start FruityProxy, BDFproxy and MITMf. ( I have tried to activate everyone of them, one-by-one) But they fail.
This is why would be easier to have the debug mode or aggregated logs. =]

LucaBongiorni, 2015-07-21:
UPDATE: If I run hostapd, I can load sucesfully FruityProxy.
Now I cannot figure out (yet) why FruityProxy & co. do not run with mana or karma. =]

xtr4nge, 2015-07-21:
Hi LucaBongiorni,
Sorry, I misunderstood.

There is not an easy way of debug them all at the same time, as each module is built differently, but it is easy to debug them one by one on the console.

FruityProxy, MITMf, and BDFproxy, more or less relies on the same dependencies.

+ FruityProxy: Start the AP mode that you want to use, and go to `/usr/share/fruitywifi/www/modules/fruityproxy/includes/FruityProxy-master`, then run `./fruityproxy.py`

Note: this will not give you a working FruityProxy module as the traffic on port 80 needs to be redirected to port 8080. If you create an iptables rule (80 -> 8080), that will give you a working FruityProxy module. You can see the execution steps on `module_action.php` on `/usr/share/fruitywifi/www/modules/fruityproxy/includes/`, but what is important on this steps is that you can see what happens when you exec `./fruityproxy.py`.

+ MITMf: Same as before, but MITMf requires the interface as parameter: `./mitmf.py -i wlan0`. (check the iptable rule on module_action.php)

Let me know.
regards,

LucaBongiorni, 2015-07-22:
Thanks!
I will start deep debugging.
P.S. If I would like to change default 8080 port of fruityproxy... would be a big mess?
Asking because I need that 8080 for other stuff ;]

xtr4nge, 2015-08-30:
Hi LucaBongiorni,
I will add the option to change the port as soon as possible.

For further questions (except for bugs) please open a thread in the forum http://fruitywifi.boards.net/

regards ;)

closed 2015-08-30 by xtr4nge