xtr4nge/FruityWifi #148: Few bug captive

larogtek12, 2015-07-29:
www.captive/includes/iptables
IPTABLES=/sbin/iptables
IFACE_INTERNET="wlan0"
IFACE_WIFI="wlan2"
___________________________________________
www.captive/index.php
was
preg_match('/..:..:..:..:..:../',$mac , $matches);
maybe 
preg_match('/..:..:..:..:..:../',$mac[0] , $matches);
______________________________________________
www.captive/process.php

    $exec = "-I internet 1 -t mangle -m mac --mac-source $mac -j RETURN";
    //exec("$bin_danger \"$exec\""); //DEPRECATED
    //exec_fruitywifi($exec); - this don't work, 
   shell_exec('sudo iptables ' . $exec); - and I do that, it's work

I use Kali Linux last update

xtr4nge, 2015-08-08:
Hi larogtek12,
Thanks a lot for reporting this issue. I will take a look.

regards,

xtr4nge, 2015-08-09:
Hi larogtek12,
Captive has been updated and fixed.

Thanks and regards,

closed 2015-08-09 by xtr4nge