xtr4nge/FruityWifi #151: Install script is not working with raspbian as dnsmasq takes over during install

aleskr, 2015-08-01:
I just tried to install v2.2 on a Pi2 following the simple stems in the tutorial, but during the install it will break as dnsmasq starts and stops all DNS-queries form working. I had to disable dnsmasq and rerun the install for it to complete.

xtr4nge, 2015-08-08:
Hi aleskr,
Thanks a lot for reporting this. I will add some code preventing this problem on R-Pi.

regards,

Keldorb, 2015-08-28:
Ive tried to install it by git clone.
It worked fine for me. Every other install method didnt went good for my raspberry.
But i face the problem that i cant login on the webpage with admin:admin....

@aleskr just try 
sudo git clone https://github.com/xtr4nge/FruityWifi.git
sudo sh <path_to_fruitywifi>/install-FruityWifi.sh

bandwagonz, 2015-09-03:
@Keldorb I'm having the same issue.  Running Kali 1.1.0 on RPI2

xtr4nge, 2015-09-03:
Hi guys,
I will need more details and screenshots to help you out with this problem, as I don't know how your problem looks like.
Did you checked the FAQ? https://github.com/xtr4nge/FruityWifi/wiki/FAQ

There is a thread on the forum related with RPi and Kali Linux: http://fruitywifi.boards.net/thread/54/install-raspberry-running-kali-linux

regards,

bandwagonz, 2015-09-04:
@xtr4nge I'll create a different ticket for my issue as I didn't fully read this one when I posted my reply.

OranJuiceJohnes, 2015-09-16:
Just to add some Data. I installed a brand new Raspian Image on my Raspi2. Updated everything. I is on wheezy. 

git clone and installed as well as install for the modules. I did this with sudo as the user pi, so I had to change some of the permissions afterwards, but I guess that was my fault.

It is up and running. 

I have a couple of questions and problems with some of the modules, but let's just call that inexperience with fruity.

xtr4nge, 2015-09-30:
Hi OranJuiceJohnes,
Thanks a lot for the info.

please reference the forum for questions about modules if there is not a bug:
http://fruitywifi.boards.net/board/4/fruitywifi-modules

regards,

closed 2015-09-30 by xtr4nge