xtr4nge/FruityWifi #152: how to use mimtf?

twayz, 2015-08-03:
Hi,

Thank you so much for adding so many different modules. I am not able to get the sslstrip2 module to work correctly whether I enable it from within mimtf or by itself. It correctly redirects https:// domains. However, the page it redirects it to is always unavailable. But, when I manually run mimtf.py from the modules folder without starting FruityWifi, it correctly redirects https:// domains. I was wondering if maybe I am doing something wrong? I usually use these options when I run mimtf.py manually:

./mitmf.py -i eth0 --spoof --arp --hsts --gateway 192.168.11.1 --target 192.168.11.xx --log debug

and it works. I also have the hsts plugin enabled from the fruitywifi admin page. Is there anything specific I have to do in order to get mimtf working from fruitywifi admin page?

Thank you so much!
Twayz

xtr4nge, 2015-08-03:
Hi twayz,
`MITMf` core is SSLstrip2 so you don't need to start `sslstrip2` module, just start MITMf.

regards,

xtr4nge, 2015-08-30:
Please reopen the topic in the forum if required: http://fruitywifi.boards.net/

regards,

closed 2015-08-30 by xtr4nge