xtr4nge/FruityWifi #153: Can't change settings or start modules

ALanR917, 2015-08-13:
Hi, I recently installed FruityWifi, and I am having the hardest time setting it up. I want to use an ethernet port to supply the internet, and a wireless card to host the rogue AP. My problem is that I cant seem to change any settings or start any modules. It refreshes the page whenever I change a setting and then it goes right back to what it was before. What can I do to solve this problem? Thank you!

xtr4nge, 2015-08-13:
Hi ALanR917
I hope you are well.

It looks like a permission problem and/or the wireless card is not supported.
Please check the FAQ on the forum: http://fruitywifi.boards.net/thread/52/faq

For reference: https://github.com/xtr4nge/FruityWifi/issues/137

Let me know if that solved the problem,
regards,

xtr4nge, 2015-08-30:
Please re-open this thread in the forum if required: http://fruitywifi.boards.net/

regards,

closed 2015-08-30 by xtr4nge