xtr4nge/FruityWifi #158: OSX support ?

donstefan, 2015-09-04:
Is there any tutorial on how to install on osx ? Thanks

daforcebreda, 2015-09-04:
The application can be installed in any Debian based system. Tested in
Debian, Kali Linux, BugTraq, Kali Linux ARM (Raspberry Pi), Raspbian
(Raspberry Pi), Pwnpi (Raspberry Pi).

I Think OSX is not like a Debian based system. mabye you can install a
VM of Debian in OSX.

xtr4nge, 2015-09-04:
Hi donstefan,
FruityWiFi is being developed for Debian based systems.
Could be possible to port it, but you need to install and/or compile the
dependencies of each module and also you need to fix and change a lot of
paths on modules and core system. (It will be a pain)

I use and develop FruityWiFi (mainly) on OSX using VMs. I have better
results on VMware Fusion, but VirtualBox works fine with some USB dongles.

Regards

closed 2015-09-05 by xtr4nge