xtr4nge/FruityWifi #159: Unable to Log in With admin:admin

bandwagonz, 2015-09-04:
Running a fresh install of Kali 1.1.0 on RPI 2

Fruity WiFi was the only thing I manually installed after writing the Kali image to an SD card.

Did a got clone on the master branch repo, then executed both .sh scripts to install all the modules and deps. All installed correctly with no error output 

When I tried logging in to the web interface using admin as the username and pass, it didn't work. I tried this multiple times and sometimes I'd get the "Who are you" message in the username field.

I checked here first and the FAQ. I ended up enabling short tags in php, then restarted everything and I still wasn't able to log in. As a last ditch effort I tried changing the username and password in one of the www configs but that didn't work either. 

I don't think I missed something is there any way for me to diagnose it more to figure out the issue?

xtr4nge, 2015-09-05:
Hi bandwagonz,
I just installed FruityWiFi from Kali Linux repositories and I didn't have any problem.

Links
http://images.kali.org/kali-2.0.1-rpi.img.xz
http://images.kali.org/kali-2.0.1-rpi2.img.xz

Example [core]:
`apt-get install fruitywifi-core`

`/etc/init.d/fruitywifi restart`
`/etc/init.d/php5-fpm restart`

Example [module-mana]:
`apt-get install fruitywifi-module-mana`

all: 
`apt-get install fruitywifi`

`/etc/init.d/fruitywifi restart`
`/etc/init.d/php5-fpm restart`

Please check the forum.

regards,

bandwagonz, 2015-09-08:
I'll try with Kali 2.0, so far I haven't had the best of luck with this distro on my laptop, which is why I opted for going with 1.1.0 on the RPI2 and using git to get the latest version. 

I'll start from a fresh install tonight and report back.

jaredmcqueen, 2015-09-15:
this also happened to me after installing fruityproxy.  In chrome's javascript debugger:

Uncaught Error: Bootstrap's JavaScript requires jQuery(anonymous function) @ bootstrap.js:6
bootstrap-switch.js:669 Uncaught TypeError: Cannot read property 'fn' of undefined(anonymous function) @ bootstrap-switch.js:669(anonymous function) @ bootstrap-switch.js:708(anonymous function) @ bootstrap-switch.js:710
main.js:6 Uncaught ReferenceError: $ is not defined

line 6 of main.js is the culprit it looks like:
  $(function() {

xtr4nge, 2015-10-14:
Hi, please check the thread related on the forum
http://fruitywifi.boards.net/thread/72/rpi-login-issue

regards,

closed 2015-10-14 by xtr4nge