xtr4nge/FruityWifi #160: Rpi2 + AR9271 Can see AP for about 10sec

vsboost, 2015-09-07:
Hi All,

Installed Raspbian and Fruity 2.2 today running an Alfa AWUS036NHA

Problem is when i start the AP i can see the ssid being transmitted from another computer for around 10sec and vanishes.

Sometimes when i press stop for the AP i'm unable to access the fruity web interface i get a 504 gateway timeout error. If im logged in via SSH running the ifconfig command crashes and i have to reboot the Pi from the power cord.

xtr4nge, 2015-09-07:
Hi vsboost,
Could be a power issue if you are not using a powered USB hub
I noticed similar behaviour with the Alfa AWUS036NHA and other usb dongles without using a powered USB hub on the RPi.

Please open/check threats in the forum with this topic: http://fruitywifi.boards.net

I want to keep GitHub Issues to track bugs only.

regards,

closed 2015-09-07 by xtr4nge

vsboost, 2015-09-08:
Thanks, changed the wifi card to a TP-Link and works.