xtr4nge/FruityWifi #161: mana not starting karma ok

vsboost, 2015-09-08:
Tried to run mana but wont start. below is the log file.  karma starts no problem.

hostapd_free_hapd_data: Interface wlan1 wasn't started
nl80211 driver initialization failed.
nl80211: Could not configure driver mode
Configuration file: /usr/share/fruitywifi/www/modules/mana/includes/conf/hostapd.conf
hostapd_free_hapd_data: Interface wlan1 wasn't started
nl80211 driver initialization failed.
nl80211: Could not configure driver mode
Configuration file: /usr/share/fruitywifi/www/modules/mana/includes/conf/hostapd.conf

vsboost, 2015-09-08:
this is the error when starting Hostapd-Mana under config

Using interface wlan1 with hwaddr xx:xx:xx:xx:xx:xx and ssid 'xxxxxxxxx'
Failed to update rate sets in kernel module
Configuration file: /usr/share/fruitywifi/www/modules/karma/includes/conf/hostapd.conf

vsboost, 2015-09-08:
Sorry forgot chipset is: Atheros Communications, Inc. AR9271 802.11n

vsboost, 2015-09-08:
Wont start a new one but noticed this error in karma, again Hostapd mode works fine.

Configuration file: /usr/share/fruitywifi/www/modules/karma/includes/conf/hostapd.conf
Failed to update rate sets in kernel module
Using interface wlan1 with hwaddr xxxxxxxxxx and ssid 'xxxxxxxx'

vsboost, 2015-09-13:
update, hostapd-karma / karma are now working, not sure what happened but i edited /etc/NetworkManager/NetworkManager.conf

[keyfile]
unmanaged-devices=mac:[mac address of interface to exclude]

rebooted it and the module started, strange thing is when i look at the same file it now looks like this.

[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

So not too sure whats happened here but karma works, but unfortunately mana still cant start

xtr4nge, 2015-09-13:
Hi vsboost,
I will re-check this as I added code on previous version to avoid this problem with NetwrokManager, but maybe I messed up on v2.2. 

I will take a look.
regards,

vsboost, 2015-09-14:
just to let you know i installed kali on the pi2 and did an apt-get install fruitywifi and was able to get mana started.

Thanks for your help.

xtr4nge, 2015-09-30:
No prob, 
regards,

closed 2015-09-30 by xtr4nge