xtr4nge/FruityWifi #166: wlan0mon not detected

grayburn, 2015-10-28:
Running Fruitywifi 2.2 on a Raspberry Pi 2 with Kali 2.0. Latest version of aircrack-ng does not use mon0 anymore. Under config >> additional interfaces, I select my wlan0 device (TP-Link WN722N) and click start. On the terminal, iwconfig shows my wlan0mon device but FruityWifi does not show it. I am assuming it is expecting mon0.

xtr4nge, 2016-01-01:
Hi grayburn,
Thanks for the info, I will take a look ;)

regards,

chilcano, 2016-01-02:
Hi there, 
I am facing the same problem. I have fixed by modifying `/usr/share/fruitywifi/www/functions.php` as follow:

```bash
...
function start_monitor_mode($iface) {

    $bin_danger = "/usr/share/fruitywifi/bin/danger";

    // START MONITOR MODE (mon0)
    $iface_mon0 = exec("/sbin/ifconfig |grep mon0");
    if ($iface_mon0 == "") {
        ////$exec = "/usr/bin/sudo /usr/sbin/airmon-ng start $iface";
        //exec("$bin_danger \"" . $exec . "\"", $output); //DEPRECATED
        ////exec_fruitywifi($exec);

        // ****
        $exec = "/usr/bin/sudo iw phy phy0 interface add mon0 type monitor";
        exec_fruitywifi($exec);
        $exec = "/usr/bin/sudo iw dev wlan0 del";
        exec_fruitywifi($exec);
        $exec = "/usr/bin/sudo /sbin/ifconfig mon0 down";
        exec_fruitywifi($exec);
        $exec = "/usr/bin/sudo /sbin/iwconfig mon0 mode monitor";
        exec_fruitywifi($exec);
        $exec = "/usr/bin/sudo /sbin/ifconfig mon0 up";
        exec_fruitywifi($exec);

     }

}

function stop_monitor_mode($iface) {

    $bin_danger = "/usr/share/fruitywifi/bin/danger";

    // START MONITOR MODE (mon0)
    $iface_mon0 = exec("/sbin/ifconfig |grep mon0");
    if ($iface_mon0 != "") {
        ////$exec = "/usr/bin/sudo /usr/sbin/airmon-ng stop mon0";
        //exec("$bin_danger \"" . $exec . "\"", $output); //DEPRECATED
        ////exec_fruitywifi($exec);

        // ****
        $exec = "/usr/bin/sudo iw dev mon0 del";
        exec_fruitywifi($exec);
        $exec = "/usr/bin/sudo iw phy phy0 interface add wlan0 type managed";
        exec_fruitywifi($exec);
        $exec = "/usr/bin/sudo /sbin/ifconfig wlan0 up";
        exec_fruitywifi($exec);

    }

}
...
```

Regards.

c0mix, 2016-01-02:
i had the same problem, just search mon0 in this repo, open the interested files and change "...grep mon0..." with "... grep 'mon0 | wlan0mon'"

chilcano, 2016-01-02:
Nice trick c0mix !

In my case (exopc + kali linux) `airmon-ng start wlan0` doesn't work for 2 reasons:
* `lsusb` isn't installed
  * `$ sudo apt-get install usbutils`
* `airmon-ng` creates `wlan0mon` instead of `mon0`

regards.

xtr4nge, 2016-01-31:
Hi guys,
Fixed on FruityWiFi v2.3 without using airmon-ng. I will keep the monitor interface as mon0 for the time being.

regards,

closed 2016-01-31 by xtr4nge

nadirchine, 2016-02-01:
@Chicano ! I have the same problem with airmon-ng ... There is only wlan0mon

xtr4nge, 2016-02-01:
Hi nadirchine,
You need to upgrade to FruityWiFi v2.3 to fix the problem on `Additional Interfaces` 
(mon0 vs wlan0mon issue)

regards,

nadirchine, 2016-02-01:
thankyou very much, i  will upgrade it

https://github.com/xtr4nge/FruityWifi/issues/166#issuecomment-178163381

whywant, 2019-06-06:
Looking for my friend：Medjitena Nadir，Who has his email address? Please tell me，Thanks